rogerclarkmelbourne/Arduino_STM32 #190: Wrong values for ram.maximum_size in boards.txt of F1 branch?!

edogaldo, 2016-07-19:
Dear Roger, I think following lines in F1's boards.txt are wrong, could you please check?

mapleRET6.upload.ram.maximum_size=492152
genericSTM32F103V.menu.device_variant.STM32F103VC.upload.ram.maximum_size=492152
genericSTM32F103Z.menu.device_variant.STM32F103ZC.upload.ram.maximum_size=492152

the correct value should be 49152.

Best, E.

rogerclarkmelbourne, 2016-07-19:
It definitely looks like a mistake / typo

I will double check the correct value and then update it

rogerclarkmelbourne, 2016-07-19:
http://www.st.com/content/st_com/en/products/microcontrollers/stm32-32-bit-arm-cortex-mcus/stm32f1-series/stm32f103.html?querycriteria=productId=LN1565

Agreed

xC have 48k 
xE have 64k 

I don't think I bothered with the F or G series as boards with that MCU on it, don't seem to be widely available.

rogerclarkmelbourne, 2016-07-19:
Should be fixed by
https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/18f22333578abf995a3a084847dec2b1d0cd6e97

edogaldo, 2016-07-20:
HI Roger, as for "xE have 64k", shouldn't it be:

mapleRET6.upload.ram.maximum_size=65536

?

Best, E.

edogaldo, 2016-07-20:
Hi Roger, I'm checking the Maple boards configuration and I have some more questions:

- Is it correct that, considering the original bootloader, "Maple (Rev 3)" has lower boundaries than "Maple Mini"?
```
maple.upload.ram.maximum_size=17000
maple.upload.flash.maximum_size=108000
maple.upload.maximum_size=108000

mapleMini.menu.bootloader_version.original.upload.ram.maximum_size=17408
mapleMini.menu.bootloader_version.original.upload.flash.maximum_size=110592
mapleMini.menu.bootloader_version.original.upload.maximum_size=110592
```

- Maple RET6 is rated for 64k ram and 512k flash so, even considering the bootloader size, its boundaries seem quite undersized, isn't it?
```
mapleRET6.upload.ram.maximum_size=49152
mapleRET6.upload.flash.maximum_size=262144
mapleRET6.upload.maximum_size=262144
```

Best, E.

rogerclarkmelbourne, 2016-07-20:
Yes
Sorry this i my mistake

I will correct

rogerclarkmelbourne, 2016-07-20:
Edit

Actually, I think I inherited some incorrect numbers  and no one, including me, has noticed.

edogaldo, 2016-07-20:
This confirms those boards are not so popular.. :)

rogerclarkmelbourne, 2016-07-20:
The Maple board is presumed to have the old / original bootloader which reserves 3K of RAM.

So the value should be 17408 instead of 17000 and the bootloader took up more space (but the Flash value is wrong as well, it should be the same as the Maple mini with original bootloader

Whoever wrote the original values, didnt realise that 1k is 1024 bytes. I remember changing a lot of the values when I inherited the code from bob cousins, but missed some of the boards.

rogerclarkmelbourne, 2016-07-20:
The original bootloader took 12 k flash and 3 k ram, So I think this gives

mapleMini.menu.bootloader_version.original.upload.flash.maximum_size=118784
mapleMini.menu.bootloader_version.original.upload.ram.maximum_size=17408

maple.upload.ram.maximum_size=17408
maple.upload.flash.maximum_size=118784

mapleRET6.upload.ram.maximum_size=62464
mapleRET6.upload.flash.maximum_size=249856

rogerclarkmelbourne, 2016-07-20:
Looks like these are wrong as well

nucleo_f103rb.upload.ram.maximum_size=17000
nucleo_f103rb.upload.flash.maximum_size=108000

As this board does not normally use a bootloader as it had STLink attached

So I think it should be

nucleo_f103rb.upload.ram.maximum_size=20480
nucleo_f103rb.upload.flash.maximum_size=131072

Also

I should add

genericSTM32F103C.menu.device_variant.STM32F103CB.upload.maximum_size=131072
genericSTM32F103C.menu.device_variant.STM32F103CB.upload.ram.maximum_size=20480
genericSTM32F103C.menu.device_variant.STM32F103CB.upload.flash.maximum_size=131072

There are some other things which are technically an error but which the IDE won't let me fix, e,g. F103C8 and CB are the same board but have a "variant" to select the C8 or CB

But it also has different upload methods, one of which is the bootloader, which would mean that the flash size should be changed if its selected, but the menu's are not cascading so there isn't a mechanism allow me to take both menus into consideration when determining the flash size.

(Same applies to R V and Z series generic boards)

edogaldo, 2016-07-20:
12k flash the original bootload?! Huh, I thought less.
for mapleRET6 I would say:
`mapleRET6.upload.maximum_size=512003`

rogerclarkmelbourne, 2016-07-20:
Yes.

You are right again, 

If RE has 512k and bootloader takes 12k 

bytes = 500 * 1024 -= 512000 

??

rogerclarkmelbourne, 2016-07-20:
Umm
let me check
I think I'm wrong

rogerclarkmelbourne, 2016-07-20:
arrrg

OK.

The original bootloader took 0x5000 = 20k

The new bootloader takes 0x2000 = 8k

I got confused, I remember now the new bootloader is 12k smaller (not that the old bootloader was 12k )

I'll need to re-do the maths 

So the original maple is OK

edogaldo, 2016-07-20:
(128-20)*1024=110592
(512-20)*1024=503808

edogaldo, 2016-07-20:
for ram:
(20-3) x 1024 = 17408
(64-3) x 1024 = 62464

edogaldo, 2016-07-20:
I think MipleMini is OK, need review Maple v3 and Maple RET6 (and maybe also others.

edogaldo, 2016-07-20:
> But it also has different upload methods, one of which is the bootloader, which would mean that the flash size should be changed if its selected, but the menu's are not cascading so there isn't a mechanism allow me to take both menus into consideration when determining the flash size.

Maybe for the generic boards the "maximum_size" should better go into the upload definitions

edogaldo, 2016-07-20:
Here a proposal for generic V boards:

```
########################### Generic STM32F103V with Bootloader ###########################

genericSTM32F103V_bl.name=Generic STM32F103V series with Bootloader
genericSTM32F103V_bl.vid.0=0x1EAF
genericSTM32F103V_bl.pid.0=0x0004
genericSTM32F103V_bl.build.variant=generic_stm32f103v
genericSTM32F103V_bl.build.vect=VECT_TAB_ADDR=0x8002000
genericSTM32F103V_bl.build.core=maple
genericSTM32F103V_bl.build.board=GENERIC_STM32F103V
genericSTM32F103V_bl.build.upload_flags=-DSERIAL_USB -DGENERIC_BOOTLOADER
genericSTM32F103V_bl.build.ldscript=ld/stm32f103veDFU.ld
genericSTM32F103V_bl.upload.use_1200bps_touch=false
genericSTM32F103V_bl.upload.file_type=bin
genericSTM32F103V_bl.upload.auto_reset=true

genericSTM32F103V_bl.build.error_led_port=GPIOE
genericSTM32F103V_bl.build.error_led_pin=6

genericSTM32F103V_bl.upload.protocol=maple_dfu
genericSTM32F103V_bl.upload.tool=maple_upload
genericSTM32F103V_bl.upload.usbID=1EAF:0003
genericSTM32F103V_bl.upload.altID=2

#---------------------------- VARIANTS ---------------------------

genericSTM32F103V_bl.menu.device_variant.STM32F103VC=STM32F103VC
genericSTM32F103V_bl.menu.device_variant.STM32F103VC.build.cpu_flags=-DMCU_STM32F103VC -DUSART_RX_BUF_SIZE=96 -DUSART_TX_BUF_SIZE=96
genericSTM32F103V_bl.menu.device_variant.STM32F103VC.upload.maximum_size=253952
genericSTM32F103V_bl.menu.device_variant.STM32F103VC.upload.ram.maximum_size=49152
genericSTM32F103V_bl.menu.device_variant.STM32F103VC.upload.flash.maximum_size=253952
genericSTM32F103V_bl.menu.device_variant.STM32F103VC.build.ldscript=ld/stm32f103vc.ld

genericSTM32F103V_bl.menu.device_variant.STM32F103VD=STM32F103VD
genericSTM32F103V_bl.menu.device_variant.STM32F103VD.build.cpu_flags=-DMCU_STM32F103VD -DUSART_RX_BUF_SIZE=128 -DUSART_TX_BUF_SIZE=128
genericSTM32F103V_bl.menu.device_variant.STM32F103VD.upload.maximum_size=385024
genericSTM32F103V_bl.menu.device_variant.STM32F103VD.upload.ram.maximum_size=65536
genericSTM32F103V_bl.menu.device_variant.STM32F103VD.upload.flash.maximum_size=385024
genericSTM32F103V_bl.menu.device_variant.STM32F103VD.build.ldscript=ld/stm32f103vd.ld

genericSTM32F103V_bl.menu.device_variant.STM32F103VE=STM32F103VE
genericSTM32F103V_bl.menu.device_variant.STM32F103VE.build.cpu_flags=-DMCU_STM32F103VE -DUSART_RX_BUF_SIZE=128 -DUSART_TX_BUF_SIZE=128
genericSTM32F103V_bl.menu.device_variant.STM32F103VE.upload.maximum_size=516096
genericSTM32F103V_bl.menu.device_variant.STM32F103VE.upload.ram.maximum_size=65536
genericSTM32F103V_bl.menu.device_variant.STM32F103VE.upload.flash.maximum_size=516096
genericSTM32F103V_bl.menu.device_variant.STM32F103VE.build.ldscript=ld/stm32f103ve.ld

########################### Generic STM32F103V w/o Bootloader ###########################

genericSTM32F103V.name=Generic STM32F103V series w/o Bootloader
genericSTM32F103V.vid.0=0x1EAF
genericSTM32F103V.pid.0=0x0004
genericSTM32F103V.build.variant=generic_stm32f103v
genericSTM32F103V.build.vect=VECT_TAB_ADDR=0x8000000
genericSTM32F103V.build.core=maple
genericSTM32F103V.build.board=GENERIC_STM32F103V
genericSTM32F103V.upload.use_1200bps_touch=false
genericSTM32F103V.upload.file_type=bin
genericSTM32F103V.upload.auto_reset=true

genericSTM32F103V.build.error_led_port=GPIOE
genericSTM32F103V.build.error_led_pin=6

#---------------------------- VARIANTS ---------------------------

genericSTM32F103V.menu.device_variant.STM32F103VC=STM32F103VC
genericSTM32F103V.menu.device_variant.STM32F103VC.build.cpu_flags=-DMCU_STM32F103VC -DUSART_RX_BUF_SIZE=96 -DUSART_TX_BUF_SIZE=96
genericSTM32F103V.menu.device_variant.STM32F103VC.upload.maximum_size=262144
genericSTM32F103V.menu.device_variant.STM32F103VC.upload.ram.maximum_size=49152
genericSTM32F103V.menu.device_variant.STM32F103VC.upload.flash.maximum_size=262144
genericSTM32F103V.menu.device_variant.STM32F103VC.build.ldscript=ld/stm32f103vc.ld

genericSTM32F103V.menu.device_variant.STM32F103VD=STM32F103VD
genericSTM32F103V.menu.device_variant.STM32F103VD.build.cpu_flags=-DMCU_STM32F103VD -DUSART_RX_BUF_SIZE=128 -DUSART_TX_BUF_SIZE=128
genericSTM32F103V.menu.device_variant.STM32F103VD.upload.maximum_size=393216
genericSTM32F103V.menu.device_variant.STM32F103VD.upload.ram.maximum_size=65536
genericSTM32F103V.menu.device_variant.STM32F103VD.upload.flash.maximum_size=393216
genericSTM32F103V.menu.device_variant.STM32F103VD.build.ldscript=ld/stm32f103vd.ld

genericSTM32F103V.menu.device_variant.STM32F103VE=STM32F103VE
genericSTM32F103V.menu.device_variant.STM32F103VE.build.cpu_flags=-DMCU_STM32F103VE -DUSART_RX_BUF_SIZE=128 -DUSART_TX_BUF_SIZE=128
genericSTM32F103V.menu.device_variant.STM32F103VE.upload.maximum_size=524288
genericSTM32F103V.menu.device_variant.STM32F103VE.upload.ram.maximum_size=65536
genericSTM32F103V.menu.device_variant.STM32F103VE.upload.flash.maximum_size=524288
genericSTM32F103V.menu.device_variant.STM32F103VE.build.ldscript=ld/stm32f103ve.ld

#---------------------------- UPLOAD METHODS ---------------------------

genericSTM32F103V.menu.upload_method.serialMethod=Serial
genericSTM32F103V.menu.upload_method.serialMethod.upload.protocol=maple_serial
genericSTM32F103V.menu.upload_method.serialMethod.upload.tool=serial_upload

genericSTM32F103V.menu.upload_method.STLinkMethod=STLink
genericSTM32F103V.menu.upload_method.STLinkMethod.upload.protocol=STLink
genericSTM32F103V.menu.upload_method.STLinkMethod.upload.tool=stlink_upload
genericSTM32F103V.menu.upload_method.STLinkMethod.build.upload_flags=-DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1 -DSERIAL_USB -DGENERIC_BOOTLOADER

genericSTM32F103V.menu.upload_method.BMPMethod=BMP (Black Magic Probe)
genericSTM32F103V.menu.upload_method.BMPMethod.upload.protocol=gdb_bmp
genericSTM32F103V.menu.upload_method.BMPMethod.upload.tool=bmp_upload
genericSTM32F103V.menu.upload_method.BMPMethod.build.upload_flags=-DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG
```

rogerclarkmelbourne, 2016-11-10:
I think this is partially fixed, but there are still some mistakes that need to be rectified

stevstrong, 2019-09-16:
I cannot see any mistake anymore, so we should maybe close this?

rogerclarkmelbourne, 2020-02-06:
I think this has been fixed a long time ago

closed 2020-02-06 by rogerclarkmelbourne