rogerclarkmelbourne/Arduino_STM32 #191: Seeed Arch Max Support

sbrown4, 2016-07-20:
Attached for your review are two patches that add this board. The first just adds the board and the second enables USART6. That's the usart hooked to the interface mcu and appears as the ttyACM device. The "Serial" device is now assigned to this usart. Serial5 is USART1. That change only affects this board.

Flashing is done by copying the .bin file to the MBED disk. This has been tested on Linux, but the mbed_upload.bat file for windows has not.

Hope this the right place to submit these.
Steve

[0001-Add-Seeed-Arch-Max-board-as-MBED-device.patch.txt](https://github.com/rogerclarkmelbourne/Arduino_STM32/files/373393/0001-Add-Seeed-Arch-Max-board-as-MBED-device.patch.txt)

[0002-Enable-USART6-on-the-Seeed-Arch-Max-board-as-Serial.patch.txt](https://github.com/rogerclarkmelbourne/Arduino_STM32/files/373396/0002-Enable-USART6-on-the-Seeed-Arch-Max-board-as-Serial.patch.txt)

rogerclarkmelbourne, 2016-07-21:
Any chance you can do this as a Pull Request?

sbrown4, 2016-07-23:
OK, I'll try.

rogerclarkmelbourne, 2016-11-10:
Closed due to inactivity

closed 2016-11-10 by rogerclarkmelbourne

stevstrong, 2016-11-10:
I should finally take out my Seed Arch board from the drawer and find a useful application for it...

rogerclarkmelbourne, 2016-11-10:
Yep. I have one of those as well, but no use for it ;-)

stevstrong, 2017-07-14:
I will soon try to add support for Seeed Arch Max into my repo.
This is how to apply a patch when the patch file is available:
`git apply issue1.patch`