rogerclarkmelbourne/Arduino_STM32 #193: Bootloader description not complete?

stevstrong, 2016-07-25:
Hi Roger,
I have the impression that the description of the bootloader under:
https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Bootloader
is not complete, it ends before pointing to the stm32duino bootloader.
Please check.

rogerclarkmelbourne, 2017-05-25:
I don't think there has been anything deleted from the bootloader page. 

I looked at the page history 

https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Bootloader/_history 

and its mostly been modified by @victorpv 

I must admit, I"m not sure why GitHub seems to let anyone other than me edit the wiki, but I don't think thats a problem.

If you think there is something missing, can you try editing the page yourself and let me know whether that works.

stevstrong, 2017-05-25:
Hi Roger,
I managed to edit the page, so I will close this issue.

closed 2017-05-25 by stevstrong