rogerclarkmelbourne/Arduino_STM32 #196: Always updatable libraries in Arduino IDE 1.6.10

per1234, 2016-07-29:
1. Install Arduino IDE 1.6.10
2. Install Arduino_STM32 core
3. Start Arduino IDE
4. Tools > Board > Maple Mini(or any other STM32F1 board)
5. Sketch > Include Library > Manage Libraries... > Type > Updatable
6. Update the Servo and FreeRTOS libraries - Instead of disappearing from the Updatable list as expected, they remain in the list no matter what you do. If enabled, the Arduino IDE will continue to show Updatable library notifications when a STM32F1 board is selected.

The solution is to add library.properties files to these libraries. I recommend creating unique name values, otherwise the issue will still occur when there is a greater version of the library of the same name available via Library Manager, for example:
```
name=Servo(STM32F1)
```

Isaac96, 2016-07-29:
Hi perl :smiley:

rogerclarkmelbourne, 2016-07-29:
IDE 1.6.10 is not currently supported 
The newest supported verison is 1.6.9 (see the readme)

per1234, 2016-07-29:
I understand that, I assumed that eventually you'll want to support 1.6.10 and would appreciate some help in that direction. I don't even use Arduino_STM32 but I'm familiar with this issue so I thought I'd make the effort to share my knowledge.

Try it with Arduino IDE 1.6.7, you'll see that the same issue occurs with the Servo library. The cause of the issue in 1.6.7 is a different bug, which still exists in 1.6.10, but there is a new Library Manager bug in 1.6.10 that causes it to also occur with the FreeRTOS library.

rogerclarkmelbourne, 2016-07-29:
Try using 1.6.9

per1234, 2016-07-29:
The reason it doesn't occur in 1.6.8 and 1.6.9 is because the version of the Servo library included with the Arduino IDE is the same as the latest version available on Library Manager. If they release a new version of the Servo library the always updatable Servo library issue will occur on 1.6.9. You can verify this by changing the version in the Arduino IDE 1.6.9's Servo library library.properties file to 1.1.1.

Testato, 2016-08-12:
+1

Yes thi si very important, it is not related to the IDE Version
All the modified Lbrary must have a unique name, different from the original arduino core library name.
So if you modify the original arduino Servo library, you must use a library.proprierty file, and inside it you must use a different name.

On esp github project all library is called xxx-esp8266
so you can call it Servo-STM32

You do not need change the name of the library itself, but only the name in the library.proprierty file, because this file is used for the automatic library update

rogerclarkmelbourne, 2016-11-10:
Now works fine with 1.6.12

per1234, 2016-11-10:
Still same issue for me. Did you follow my instructions to reproduce issue?
[image]

rogerclarkmelbourne, 2016-11-10:
OK

So this is not a version issue, its an issue with library names in their properties

per1234, 2016-11-11:
I do think the Arduino IDE should be able to handle this correctly and I hope that issue will be fixed in some future version but it hasn't happened yet. I've attempted to bring the issue to the developers' attention multiple times without any response. The issue was originally reported by one of the Arduino developers(https://github.com/arduino/Arduino/issues/4189) but was incorrectly closed. The issue even occurs with one of the official Arduino cores(https://github.com/arduino/Arduino/issues/5283). However, it is east enough for 3rd party cores to work around the issue by just using a unique name value in library.properties. When you think about it, this actually make sense because, for example, the Servo library included with your core is a different library from the stock Arduino Servo library and thus it's proper for it to have a different name(though the same header file name for compatibility with all sketches including that library).

rogerclarkmelbourne, 2016-11-11:
@per1234 
Yes. This is a long standing problem.

IMHO the libraries attached to a core e.g. the STM32 should take priority over the generic Arduino versions of libraries with the same name.
But for some reason the Arduino dev's don't want to do this.

I took at look in the sources and the Servo library was legacy from ages ago, and was possibly written my Leaflabs when they originally wrote LibMaple etc.
So it doesnt have a libraries properties file.

What I need to do is to copy the latest servo library properties file and the keywords file from the Arduino AVR / SAM core and then rename the library to  Servo(STM32)

BTW. I've never actually used the Servo library, but I do have some servos for testing, so I should really wire one up and check it actually works ;-)

rogerclarkmelbourne, 2016-11-12:
Hopefully fixed in this commit

https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/1892447080db8c03799e564b295d3986a0d1394e

closed 2016-11-12 by rogerclarkmelbourne