rogerclarkmelbourne/Arduino_STM32 #198: Upload method: STLink

Testato, 2016-08-12:
i tried compile the Blink examples whit all 4 Upload methods on a Generic STM32F103C8 (bluepill)

If i Verify the code whit upload method Serial and BMP the sketch is only 6KB
Instead whit STM32duino bootloader and STLink the sketch is 12KB

So i think that whit STM32duino bootloader at every upload will be inserted also the bootloader code inside, right ?

But this is not necessary for STLink, because if a people use a programmer he do not want the bootloader, and can use all the space.

rogerclarkmelbourne, 2016-08-13:
This is not a bug.

The reason the STLink is a larger binary is because it has USB CDCACM

Testato, 2016-08-13:
ok thanks, 
i removed -DSERIAL_USB flag and it return to 6KB occupied space

but why on Serial and on BMP the USB flag is not present ? isn't better have all 4 upload_method whit the same philosophy ?

For the bootloader question, i removed -DGENERIC_BOOTLOADER and the occupied space do not change, so the bootloader is always present in the board ? it is reloaded at every Upload whit all 4 upload method ?if the blink occupies 6KB how many is for bootloader ?

Philius, 2016-08-21:
Does anyone know of a bootloader that allows for device programming via USB, like the Arduino?

berryelectronics, 2016-08-21:
You mean something like the stm32duino bootloader? ;)

rogerclarkmelbourne, 2016-08-21:
https://github.com/rogerclarkmelbourne/STM32duino-bootloader

Philius, 2016-08-22:
Please forgive my utter noobiness regarding this issue.
There are no stupid questions but not looking at what other GitHub projects Roger Clark has open makes my question stand out all the more.

I'll post something in the STM32duino-bootloader about my board...

rogerclarkmelbourne, 2016-11-10:
I don't think this is an issue

closed 2016-11-10 by rogerclarkmelbourne