rogerclarkmelbourne/Arduino_STM32 #199: Arduini 1.6.9 ide warning

Testato, 2016-08-13:
when verify a code on 1.6.9 ide we receive always a warning:

Warning: platform.txt from core 'STM32 Boards (STM32duino.com)' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.

SalvorinFex, 2016-09-16:
I have also received very similar warnings from 1.6.9.
Have yet to try 1.6.11 with STM32.

rogerclarkmelbourne, 2016-09-17:
I have to leave the offending code in platform.txt otherwise 1.65 will not work.

But sooner of later I will update it, and the users still on 1.6.5 will need to upgrade

rogerclarkmelbourne, 2016-11-10:
Fixed in

https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/9096356e555ab18582f3c899a359185b500a7b8b

closed 2016-11-10 by rogerclarkmelbourne