rogerclarkmelbourne/Arduino_STM32 #200: Max RAMindication not present

Testato, 2016-08-13:
On Arduino at the end of a build, in consolle appear the used/maximum flash size and the used/maximum RAM size

On STM32duino the maximum RAM do not appear

```
Sketch uses 12,316 bytes (11%) of program storage space. Maximum is 110,592 bytes.
Global variables use 2,560 bytes of dynamic memory.
```

rogerclarkmelbourne, 2016-08-14:
Please can you suggest a solution and submit via PR

rogerclarkmelbourne, 2016-08-14:
FYI.
This appears to be a data parameter that's optional in boards.txt

upload.maximum_data_size

Introduced in 

https://github.com/arduino/Arduino/commit/7c87db3532155c051b902ac83c1feea78eb732ee#diff-4ead488680453362ffec01d74609315c

Please can you update boards.txt to add this value to all boards (including F1 and F4 repos)

Thanks

Testato, 2016-08-14:
Yes i can prepare a PR,
But i checked also on Arduino Zero and this bug is present too.
Very strange

rogerclarkmelbourne, 2016-08-14:
You could do a quick test by updating one board in boards.txt and check it works.

I'm not sure why they omitted this for the Zero, I suspect they did what we did, and used an old file which doesn't have that data in it.

It doesnt seem to be documented in the Arduino Third Party Boards specification.
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification

(I'm not surprised as Arduino don't seem to spend much time on documentation :-(  )

BTW. While I remember. I think some of the sizes defined in boards.txt are wrong, I can't remember if its just the Flash size thats wrong or whether the Ram size is wrong as well

rogerclarkmelbourne, 2016-11-10:
closed due to inactivity

Testato, 2016-12-16:
i agree that this issue have inactivity, but i think it will be a very good upgrade for the core.
It is very usefull have this info during compilation.
This is an example from the Uno
```
Sketch uses 444 bytes (1%) of program storage space. Maximum is 32,256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2,039 bytes for local variables. Maximum is 2,048 bytes.
```

So may you remain it opened ?

rogerclarkmelbourne, 2016-12-17:
@Testato 

Please send a PR for this

Testato, 2016-12-18:
@rogerclarkmelbourne 
i created 3 separated PR for the 3 board.txt correction

Testato, 2017-03-23:
Is there problems whit my PRs ?

rogerclarkmelbourne, 2017-03-23:
Not had time to test

rogerclarkmelbourne, 2019-02-27:
Merged (ages ago)

closed 2019-02-27 by rogerclarkmelbourne