rogerclarkmelbourne/Arduino_STM32 #202: Hardware-I2C broken on STM32F103ZET6

WereCatf, 2016-08-27:
Using the HWire execution grinds to a halt as soon as a call to beginTransmission(addresshere) is made, regardless of whether any device is connected to the bus or not. On bus 2 it works if no device is connected, but as soon as anything is connected to the bus the same issue happens. Tested with the HWire example-sketch and various attempts of using it in my own code.

The same code works without an issue on STM32F1C8T6.

rogerclarkmelbourne, 2016-08-27:
OK. Thats a strange one.

I've not looked at the HWire code for ages, and I accepted several PR's to fix various things in the last 6 months, but I generally only test on the BluePill or Maple mini, as testing on F103R, F103V and F103Z for every PR would take more time than I have available.

WereCatf, 2016-08-28:
I doubt anyone even expects you to test every single PR on every single device, but that means sometimes bugs slip through. And this one is quite nasty.

Interestingly, the MCU doesn't hard crash, it stalls somewhere. It still responds to dfu-reset and thus allows you to upload another sketch via the bootloader-option.

rogerclarkmelbourne, 2016-08-28:
No worries

If it hasn't hard crashed, I suppose it would be possible to debug into the HWire lib and figure out why its hanging. There are probably some loops checking the hardware reg's looking for specific bit patterns etc

WereCatf, 2016-08-28:
That goes over my skill-level, though.

rogerclarkmelbourne, 2016-08-28:
OK. No worries

I'll try to track down who did the last PR on this and I'll ask them

rogerclarkmelbourne, 2016-08-28:
I just looked at the commit history and it looks like @fergul  was the last person who submitted changes.

https://github.com/rogerclarkmelbourne/Arduino_STM32/commits/master/STM32F1/libraries/Wire/HardWire.cpp

You could try reverting the HWire files to prior to his changes, but I suspect it would be worse

You could try contacting @fergul to see if he can recommend any ways to investigate or fix the problem

WereCatf, 2016-08-28:
None of @fergul 's changes seem to affect the issue.

Has I2C ever actually worked on F103ZE? I never tried it when I got the board, I just tried it yesterday for the first time. I've been fiddling with the code and it's the call to wait_for_state_change() in cores/maple/libmaple/i2c.c that stalls and, to me, it looks like there is something wrong wrt. IRQs and/or registers themselves, but that's as far as my understanding of the code and the hardware goes.

lorneb, 2016-12-05:
Not sure if this is your same problem but I have noticed that causes of hanging whenyou select an invalid address are caused by a 0 being passed as the timeout

/master/STM32F1/libraries/Wire/HardWire.cpp in uint8 HardWire::process() 
int8 res = i2c_master_xfer(sel_hard, &itc_msg, 1, 0); //0 is the default timeout I changed to 100

This got the i2c scanner working at least

rogerclarkmelbourne, 2016-12-05:
Thanks. Perhaps @WereCatf  can try that

WereCatf, 2016-12-05:
@lorneb I'll try that later on today and report back.

WereCatf, 2016-12-08:
Nope, the change proposed by @lorneb doesn't work, I2C is still broken.

rogerclarkmelbourne, 2018-05-22:
Hardware I2C was reworked many months ago.

Please re-open this issue if you still have a problem with the latest version of the repo

closed 2018-05-22 by rogerclarkmelbourne