rogerclarkmelbourne/Arduino_STM32 #206: STM32F103TB  doesn't appear as USB device

fkpwolf, 2016-09-12:
I have a device from here http://www.hotmcu.com/stm32f103tb-arm-cortex-m3-development-board-p-222.html. It is STM32F103TBU6.

I flashed bootloader generic_boot20_pa1.bin. After device reset, I can see LED blink quickly and then slow. But no new USB device in Windows.

In Arudino, I can see a STM32F103TB menu as a "variant". So this mcu is supposed to be supported?

WereCatf, 2016-09-12:
After flashing the bootloader for the first time on my STM32F103C8T6's I had to place the bootloader in "perpetual mode" by connecting PC14 to 3.3V and resetting the board before it would appear as a USB-device. Then I flashed a blank sketch using the Arduino IDE, disconnected PC14 from 3.3V and reseted the board, after which it worked like it should.

Try placing your board in perpetual bootloader-mode and doing the same?

rogerclarkmelbourne, 2016-09-12:
Thanks @WereCatf 

I was just replying when I noticed you beat me to it.

Please register on the forum.  www.stm32duino.com as this subject was discussed recently

Note. I did briefly look at adding a new build target to the bootloader buy could not get my STM32F103T board to work at all, so I need to check if it has a fault

In theory it should be easy to add a build target to the bootloader for this board, as it has reset hardware very similar to the Maple mini, but its just on a different pin

However as my simple changes didnt work, perhaps I'm wrong about this ;-)

Anyway, its getting too late for me to look now as ist approaching 11pm

PS, This is unrelated but you should read this interesting post I just announced
http://www.stm32duino.com/viewtopic.php?f=16&t=1398

fkpwolf, 2016-09-12:
@WereCatf thanks for your quick reply. I checked but found my STM32F103TBU6(VFQFPN36) doesn't have PC14 pin.

fkpwolf, 2016-09-13:
I tried to compile the bootloader project. Modify https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/STM32F1/main.c to let device stay in "perpetual mode" forever. The hacked bootloader works as excepted on a STM32F103C8T6 device and I can see a Maple DFU device. But for STM32F103TBU6, still can't see any USB device.

So the USB lib doesn't work for this MCU or out of date? I flash "USB-Mass_Storage-MCU Flash" provided by manufacturer and it's USB function works well.

[HY-TinySTM103T-Example-(2015.06.19).zip](https://github.com/rogerclarkmelbourne/Arduino_STM32/files/469486/HY-TinySTM103T-Example-.2015.06.19.zip)

rogerclarkmelbourne, 2016-09-13:
Thanks

I had the same results that you did.

Their own example worked, but the bootloader did not appear on USB.

However someone on the stm32duino forum seems to think it works for them.

I will need to double check the forum to see what the real story is ;-)

fkpwolf, 2016-10-20:
Find workaround here: http://www.stm32duino.com/viewtopic.php?f=31&t=853&start=10#p17508.
Have to connect PA0 to GND all time.

rogerclarkmelbourne, 2016-10-20:
I dont think that workaround worked on my board.

I have a HYTiny STM32F103T http://www.hotmcu.com/stm32f103tb-arm-cortex-m3-development-board-p-222.html
which board do you have ?

fkpwolf, 2016-10-21:
ya, exactly same one. I used generic_boot20_pa1.bin.

rogerclarkmelbourne, 2016-11-10:
Should be fixed now.

I created a bootloader for this and also added a new board variant

closed 2016-11-10 by rogerclarkmelbourne