rogerclarkmelbourne/Arduino_STM32 #207: I2C/wire library: slave is not implemented

madias123, 2016-09-12:
Just for the record, because from time to time there are the same questions and troubles in the forum:
No, we do not have a working slave configuration for I2c. (onReceive()....)

rogerclarkmelbourne, 2016-10-01:
The new STM Core appeare to have I2C slave. So I think perhaps anyone needing this, will probably end up having to use that core (Albeit only when its working on boards like the BluePill)

DouglasPearless, 2017-02-25:
Has anyone (now) implemented the slave I2C function for STM32?

scargill, 2017-08-02:
A lot of info out there - some say it is in, some say it is not - yet here, in the repository, nothing since February..  Does i2c slave work?

rogerclarkmelbourne, 2017-08-02:
AFIK. No one has sent me a PR etc to add I2C slave functionality.

scargill, 2017-08-02:
Hi Roger - For those of us new to this chip - there is a lot of nonsense out there - some say it IS there, some say it isn't... any chance you can take this as a request to add i2c slave functionality?   I have a NANO as an I2c slave which can serve as a really cheap peripheral for many different devices... and a year ago I spotted the STM32 board (blue pill?) which is very cheap... it seemed this would make a better alternative - at the time people had all sorts of reasons why I2c slave functionality would not work - I was asked to look at this again today and I seem no further forward.... is it practical to make it so that it works just in the same way the WIRE library works in Arduino ? and if so - are you able to do that?

scargill, 2017-08-02:
and if so - which board are you supposed to select to get the generic stm32f103 boards to work?

rogerclarkmelbourne, 2017-08-02:
The issue is not the board. The issue is the core.

Have you looked at STM's own official Arduino core, rather than this incomplete / unofficial one

See

https://github.com/stm32duino

stevstrong, 2019-09-16:
Possible duplicate of #613 and #625 .

PR #679 (2019-10-04): WireSlave Library Rework to make I2C Slave Fully Functional
Rework the WireSlave library of the STM32F1 target, merging it with the core, and fixing all of the I2C quirks and problems.

This fixes #625 and fixes #613 and fixes #207 and possibly other open issues.

I have tested the new WireSlave code and I2C core enhancements as both a master, communicating with CAT24C256 EEPROMs and MCP4725 DACs, and as a slave, communicating with a STM32F746G Discovery board and with the [I2CDriver](https://i2cdriver.com/) test gadget (a very handy tool, BTW).

The WireSlave examples have also been updated w [... cut]
closed 2019-10-04 by stevstrong