rogerclarkmelbourne/Arduino_STM32 #208: Examples are missing from the latest repo...

SalvorinFex, 2016-09-16:
I thought I had examples of sketches for the Maple-ish boards, but when I looked I couldn't find them.
So, I took a look at an old hard drive from a defunct computer and there they were and they were dated around Dec. 2015.

IS there a particular reason why they're not included anymore?

rogerclarkmelbourne, 2016-09-17:
Do you have examples that are not in here

https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/libraries/A_STM32_Examples/examples

SalvorinFex, 2016-09-20:
That folder is listed under the folder structure, in the arduino\hardware, but I thought at some point it showed up in the actual "Examples" section in the IDE.
I have since just copied it into my Sketchbook, but then I'll need to manually check to see if any get added or updated.

rogerclarkmelbourne, 2016-09-20:
In the original LeafLabs IDE they could put their examples anywhere.

Using the official Arduino IDE, examples can only be added via libraries.

I think this may be slightly changing in IDE V1.6.11 but I do not know precisely what has been introduced in this revision

SalvorinFex, 2016-09-20:
Okay, Thanks for that info.

closed 2016-09-20 by SalvorinFex