rogerclarkmelbourne/Arduino_STM32 #209: pwmWrite not working

WereCatf, 2016-09-18:
pwmWrite() locks both of my Blue Pill - modules up when called. I tried using both Arduino IDE 1.6.9 and 1.6.11, but that doesn't seem to change anything. I do call pinMode(pin, PWM) and that works fine, but as soon as the call to pwmWrite() happens execution never progresses from that.

```
void setup(){
  Serial.begin(115200);
  pinMode(PC13, PWM);
  pwmWrite(PC13, 127);
  Serial.println("Execution doesn't reach this line.");
}

void loop(){
}
```

avikde, 2016-09-18:
It depends on which chip I am sure but pc13 isn't a timer pin on any stm32
chip I've worked with.

berryelectronics, 2016-09-18:
on the blue pill is the c8t6, right?
http://www.st.com/content/ccc/resource/technical/document/datasheet/33/d4/6f/1d/df/0b/4c/6d/CD00161566.pdf/files/CD00161566.pdf/jcr:content/translations/en.CD00161566.pdf
PC13 isnt the same as the Pin 13 on the arduino uno :)

Pins with Timer (PWMable Pins) should be:
PA1, PA2, PA3, P6, PA7, PB0, PB1, PA9, PA10, PA11, PB6, PB7, PB8, PB9
(but dont take my word, could also have some mistakes)

20:23 Uhr:

berryelectronics, 2016-09-18:
And: for these kind of problems, the http://stm32duino.com/ forum is there. 
It also has some nice Pinout diagrams (somewhere) flying around :)

WereCatf, 2016-09-18:
@berryelectronics Why do you bring Arduino Uno up? It doesn't have anything to do with this.

berryelectronics, 2016-09-18:
sorry, was kinda of an thought because the classic arduinos use pin 13 for PWM, and this is for the arduino ide, and you used pin pc13.

WereCatf, 2016-09-18:
Maybe my pinout-map is wrong, it lists PC13 as PWM-capable. As for @berryelectronics -- on the Blue Pill PC13 is connected to the built-in LED

berryelectronics, 2016-09-18:
[image]

well, there is definatly no timer on PC13, not as a normal nor as a remappable function. so you either try to use softpwm or connect a led.

WereCatf, 2016-09-18:
Aight, I need to get me a pinout that doesn't suck then. My apologies.

closed 2016-09-18 by WereCatf

berryelectronics, 2016-09-18:
the datasheet :)
or here is one from the forum:
http://reblag.dk/wordpress/wp-content/uploads/2016/07/The-Generic-STM32F103-Pinout-Diagram.pdf

jaimelaborda, 2017-04-22:
fuck! I designed a PCB using that pinout i attach and is all wrong. Do not trust that.

[image]

stevstrong, 2017-04-23:
Jaime, can you please explain what exactly is wrong with that?

jaimelaborda, 2017-04-23:
Sorry I didn't explain. Is about the PWM pins. In this pinout there are pins wrongly marked.

stevstrong, 2017-04-23:
And which ones do you mean?