rogerclarkmelbourne/Arduino_STM32 #210: Please make this available as Arduino Boards Manager package

vshymanskyy, 2016-09-20:


rogerclarkmelbourne, 2016-09-20:
@ddrown has created a boards manager package.
I'm not sure how up to date it is however.

Note.
STM have now released their own core for Arduino on STM32 including board manager package. But currently it only supports 2 boards.

https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/master/STM32/package_stm_index.json

As STM only released the code last Friday, 16/09/2016, the community has not had chance to start adding the main boards that most community members use e.g. the BluePill or Maple mini.

However we have now started to investigate what code changes are required to support the addition of all boards from the libmaple core (Arduino_STM32)

If you are interested in following what is happening with this, please register on stm32duino.com and look in the "STM Core" section

rogerclarkmelbourne, 2016-10-18:
Please use this unofficial board manager package http://dan.drown.org/stm32duino/package_STM32duino_index.json

closed 2016-10-18 by rogerclarkmelbourne