rogerclarkmelbourne/Arduino_STM32 #215: class IPAddress lacks fromString()

xymopen, 2016-10-11:
I'm try to use [WiFiEsp](https://github.com/bportaluri/WiFiEsp) with my Maple board and the IDE report missing such member function.
Request to merge this method from official board core source.

rogerclarkmelbourne, 2016-10-11:
Its not possible simply to do a "merge" as the Libmaple code base is not forked from the Arduino repo.

Please submit a Pull Request with the new functions, and it will be tested by the community and accepted if it does not cause any problems to the existing codebase.

xymopen, 2016-10-13:
Beside the AVR codes, there are also much hardward independent code serving as helpers. Those code might be merged safely.
I'll try to diff those code and find out what can be merged in a few days.

rogerclarkmelbourne, 2016-10-13:
OK

closed 2016-11-10 by rogerclarkmelbourne