rogerclarkmelbourne/Arduino_STM32 #216: Only works if I press the reset button at the exact "correct" time

chrisalbertson, 2016-10-14:
I can load sketches using the boot loader into a generic STM32F103 "blue" board.   But I have to hit the reset button at JUST to right time, seconds before the Arduino IDE starts downloading the binary file.  Then after it is loaded it hangs until I press reset once more.

Quite a lot has to be right get this far, USB has to work to a degree.   But it seems to be an issue with resets.  It may be hardware

But I think the boot loader might be fixed to wait MUCH longer for a download

Any advice on where I might look in the code or a reason why this can't be fixed.

rogerclarkmelbourne, 2016-10-14:
Please register on our community forum at www.stm32duino.com 

I recall someone posted a modified bootloader in one of the threads, but I can't remember the exact details

jaretburkett, 2016-10-18:
What OS are you using? You can also put your board in perpetual bootloader mode (dfu mode) by toggling the "Button" pin defined in the bootloader you are using immediately after reset.

rogerclarkmelbourne, 2016-11-10:
Closed due to lack of response from OP

closed 2016-11-10 by rogerclarkmelbourne