rogerclarkmelbourne/Arduino_STM32 #222: Problem using gpio PA11 while sending data over Serial1 (generic stm32f103c8t6)

fabriciopk, 2016-10-30:
I'm using the GPIO - PA11 as an enable for change the data direction of RS485 tranceiver and the Serial1(USART1) tx, rx.
The data rate is quite high (1Mbps). Looking in the chip pinout, I find that PA11 is the CTS of Serial1. It can be a problem?
In another version of the same board, I used Serial2 and PA1 for enable, and it work perfectly. 

[This is the schematic of the board that don't work.](https://github.com/fabriciopk/SEA_Firmware/blob/master/hardware/hardware-board2.pdf)

[This is the schematic of the board that work.](https://github.com/fabriciopk/SEA_Firmware/blob/master/hardware/hardware-board1.pdf)

rogerclarkmelbourne, 2016-10-30:
AFIK Flow control on USARTS is not enabled

PA11 and P12 are used for USB depending on which options you select, and there is a possibility some part of the setup code presumes they will always be used for USB.

Note. This repo is likely to be superseded by the official STM cores in the github.com/stm32duino 
But at the moment there is only support for STM Nucleo boards.

nguyenlam185, 2016-10-31:
I confront with the same issue. Please tell me how to overcome it?

rogerclarkmelbourne, 2016-10-31:
Do you mean you want to use automated flow control ?

AFIK the repo does not support USART flow control.

In terms of using PA11 and PA12 for GPIO, I've never tested this.

The repo was not initially designed to be used with PA11 and PA12 for GPIO as it was originally written by Leaflabs to support the Maple Mini board which uses PA11 and PA12 for USB.

fabriciopk, 2016-10-31:
Is there any modification I can do to use PA11 as a GPIO, wile I'm using the Serial1?
I'm using the generic STM32f103c series.

stevstrong, 2016-10-31:
As "brute force" solution, I would try to remove all USB related files from local file structure and then see where related functions are called, then remove these function calls.

fabriciopk, 2016-10-31:
Ok, now it work! I set the define BOARD_HAVE_SERIALUSB from boards.h to 0. I think I can just remove -DSERIAL_USB from boards.txt and it will work. @

fabriciopk, 2016-10-31:
And I have this warnings

/home/tiburcio/Arduino/libraries/AS5048A/AS5048A.cpp:202: warning: undefined reference to `Serial'
/tmp/build7ade03213d78c4037d2eaaf15fe93eb6.tmp/core/wirish/boards_setup.cpp.o: In function `wirish::priv::board_setup_usb()':
/home/tiburcio/arduino-1.6.9/hardware/Arduino_STM32/STM32F1/variants/generic_stm32f103c/wirish/boards_setup.cpp:95: warning: undefined reference to `Serial'

rogerclarkmelbourne, 2016-11-10:
@fabriciopk 

Whatever you are changing, you still must have SERIAL_USB defined and its attempting to use a USB Serial object that has no longer been declared.

You will not be able to remove all references to USB serial, simply by undefining stuff, as the code was never intended originally to operate without USB serial

We added in some features to allow USART1 to be Serial instead of USBSerial, but this is for a niche userbase.

I'm sure its possible to resolve this issue, but it will require more modification that you seem to be making

closed 2016-11-21 by fabriciopk

AlexanderAmelkin, 2018-07-28:
@rogerclarkmelbourne, so is there a way to enable PA11 and PA12 as GPIOs for an STM32F103VE-based board? I have such a board (Geeetech GTM32Pro) and I'm trying to use it with Marlin32 that uses Arduino_STM32 as part of PlatformIO framework. So far I wasn't able to make PA11 and PA12 work. I see your commit where you unconditionally enable `#define BOARD_HAVE_SERIALUSB`. And I don't understand how to properly disable USB initialization. My board doesn't have a USB serial (well, it does, but it's FT232 connected to USART1 of STM32). Any suggestions?

rogerclarkmelbourne, 2018-07-28:
@AlexanderAmelkin

All support is via the stm32duino community at www.stm32duino.com.

InnerBushman, 2019-08-07:
> All support is via the stm32duino community at [www.stm32duino.com](http://www.stm32duino.com).

This link doesn't work. Can it be helped here somehow? We are all stuck because the PA11/PA12/many other pins are not working/being protected. If anyone could make even some dirty hack, that could be cleaned up later, only so we could start using our boards? I would be grateful.

AlexanderAmelkin, 2019-08-22:
@InnerBushman, they work with the latest PlatformIO/ststm32 if you select generic_stm32f103ve board.
See https://github.com/platformio/platform-ststm32

stevstrong, 2019-08-22:
PA11 and PA12 can be used as GPIOs with this sequence (with enabled USB serial):
```
void setup() {
  Serial.end();
  pinMode(PA11, OUTPUT);
  pinMode(PA12, OUTPUT);
}
void loop() {
  digitalWrite(Pa11, HIGH);
  digitalWrite(PA12, LOW);
  delay(500);
  digitalWrite(PA11, LOW);
  digitalWrite(PA12, HIGH);
  delay(500);
}
```

yangkkokk, 2022-11-22:
Serial.end(); 

PC12  alone digital Write
must close output