rogerclarkmelbourne/Arduino_STM32 #226: "/bin/arm-none-eabi-g++": file does not exist

Bob0505, 2016-11-08:
Dears,

By the step as bellow.
(1)Download Arduino 1.6.12 (Windows ZIP file for non admin install)
(2)Unzip the zip file
(3)Clone this git at master branch.
(4)Arduino_STM32-master Rename to STM32
(5)Copy file STM32 into "arduino-1.6.12-windows\hardware"
(6)Open Arduino IDE edit a hello word as bellow
(7)Select the Board, Variant, Upload method.
(8)Verify/Compile the code.
(9)Arduino IDE show "exec: "/bin/arm-none-eabi-g++": file does not exist"

Test environment
    Windows 7 x64 and Windows 10 x64

======================
    Hello Word source code
======================
#define pinLED PC13

void setup() {
  Serial.begin(9600);
  pinMode(pinLED, OUTPUT);
  Serial.println("START");  
}

void loop() {
  digitalWrite(pinLED, HIGH);
  delay(1000);
  digitalWrite(pinLED, LOW);
  Serial.println("Hello World");  
}
======================

======================
    Error Msg
======================
Arduino: 1.6.12 (Windows 7), Board: "Generic STM32F103C series, STM32F103C8 (20k RAM. 64k Flash), Serial"
...
exec: "/bin/arm-none-eabi-g++": file does not exist
Error compiling for board Generic STM32F103C series.
======================

Bob0505, 2016-11-08:
I got that, Thank you sir!

https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation

Run the IDE, and on the Tools menu, select the Boards manager, and install the Arduino Zero from the list of available boards. You must do this step, it installs the arm-none-eabi-g++ toolchain!

closed 2016-11-08 by Bob0505

BlackXDragon, 2018-02-18:
Bob0505, sir... I'm not able to see anything except Arduino AVR boards in my boards manager.

eagl1, 2020-05-30:
You have to install the folder of STM32-master in the path mentioned in the main post, then you should list the stm32 boards after launching Arduino IDE.