rogerclarkmelbourne/Arduino_STM32 #231: Serial.print() doesn't work under certain circumstances

stevstrong, 2016-11-23:
[This commit](https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/1d321b6b70a0d0362fab0065ba79ad275623c58e) influenced negatively the print functionality (especially the F()macro), it should be finally fixed somehow.

stevstrong, 2016-11-29:
Apparently this solves the problem:
```
size_t Print::print(const __FlashStringHelper *ifsh)
{
  return print(reinterpret_cast<const char *>(ifsh));
}
```

rogerclarkmelbourne, 2016-11-30:
Thanks.

It seems to fix the error, so I've applied it to both the F1 and F4

closed 2016-11-30 by rogerclarkmelbourne

kc2uez, 2016-11-30:
Thank you!. I was having problems with my K3NG keyer port to the maple mini and was due to this bug. I was about to post a question on the forum, but this fixed the issue.

rogerclarkmelbourne, 2016-11-30:
No worries ;-)

commit 378cd2a6 (2016-11-30): Fixed #231