rogerclarkmelbourne/Arduino_STM32 #232: not support TIMER_INPUT_CAPTURE

sonsatthu507, 2016-11-24:
/* TIMER_INPUT_CAPTURE, TODO: In this mode, the timer can measure the pulse lengths of input signals */
i work project measure the pulse but i found in timer.h  not support TIMER_INPUT_CAPTURE

rogerclarkmelbourne, 2016-11-24:
That's correct.
There are lots of features in the STM32 hardware which are not supported by the LibMaple API.

If you would like to add support for this, please submit a PR with the code.

sonsatthu507, 2016-11-27:
Hi Roger I try the code ext interrupt pulse measurement on PA8. it's ok
`#include "HardwareTimer.h"
volatile int state = LOW; 
volatile int HighToLowCount = 0;
volatile int LowToHighCount = 0;
volatile int val;
volatile int CurrentCount;
volatile int Period;
volatile int OldCount;
volatile int PulseWidth;
volatile int OverflowCount;
volatile int timeroverflow;
 
HardwareTimer timer(1);
 
void setup()
{    
    Serial1.begin(9600);
    pinMode(PA8, INPUT);
    attachInterrupt(0, exti_handler, CHANGE);
   
    CurrentCount = 0;
    OldCount = 0;
    Period = 0;
    PulseWidth = 0;
    OverflowCount = 0;
    timeroverflow = 0;
   
        // Setup Counting Timers
    timer.setChannel1Mode(TIMER_OUTPUT_COMPARE);
    timer.pause();
    timer.setCount(0);
    timer.setPrescaleFactor(72);  //run at 72 MHz/72 = 1MHz count every 1us
    timer.setOverflow(65535);
    timer.setCompare1(1);  
    timer.attachCompare1Interrupt(timer_handler);
    timer.refresh();
    timer.resume();
 
 
}
 
void loop()
{
    Serial1.print("LowToHighCount: ");
    Serial1.print(LowToHighCount);
    Serial1.print(" HighToLowCount: ");
    Serial1.print(HighToLowCount);
    Serial1.print(" Period: ");
    Serial1.print(Period);
    Serial1.print(" PulseWidth: ");
    Serial1.print(PulseWidth);
    Serial1.print(" Overflow: ");
    Serial1.println(OverflowCount);
    delay(1000);
}
 
void exti_handler()
{
    timer.pause();
    CurrentCount = timer.getCount();
    val = digitalRead(0);
 
   
    if (val == HIGH)
    {
        LowToHighCount++;
        Period = CurrentCount;
        timer.setCount(0);
        timer.refresh();
    }
    else { // state must be LOW
        HighToLowCount++;
        PulseWidth = CurrentCount;
    }
 
    timer.resume();
}
 
void timer_handler()
{
	PulseWidth = 0;
	OverflowCount = 0;
	timeroverflow++;
}`

rogerclarkmelbourne, 2016-11-27:
OK. Thanks for sharing

rogerclarkmelbourne, 2016-11-27:
BTW.

You may like to register on www.stm32duino.com  I'm sure some other community members would be interested in your code

sonsatthu507, 2016-11-27:
thank Roger I will post my code some my project on www.stm32duino.com

closed 2016-11-27 by rogerclarkmelbourne