rogerclarkmelbourne/Arduino_STM32 #235: tone.cpp - bogus code for stm32f103c8

RickKimball, 2016-12-11:
How is that line going to work?
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/e71d176618b8b77c727af0891343a278246016fd/STM32F1/cores/maple/tone.cpp#L48

```
HardwareTimer *tone_timer = TTimer[TONE_TIMER]; // timer used to generate frequency
```

TONE_TIMER is 4 on an stm32f103c8

HardwareTimer *TTimer[4] { &TTimer1,&TTimer2,&TTimer3,&TTimer4 };

The array is only 4 items, you are indexing past the end of the array. 

When I compile it yaks:
```
home/kimballr/Arduino/hardware/stm32duino_maple/STM32F1/cores/maple/tone.cpp: In function '(static initializers for /home/kimballr/Arduino/hardware/stm32duino_maple/STM32F1/cores/maple/tone.cpp)':
/home/kimballr/Arduino/hardware/stm32duino_maple/STM32F1/cores/maple/tone.cpp:48:46: warning: array subscript is above array bounds [-Warray-bounds]
 HardwareTimer *tone_timer = TTimer[TONE_TIMER]; // timer used to generate frequency
 ```

Also the compiler is complaining about the initialization of the array itself:
```
/home/kimballr/Arduino/hardware/stm32duino_maple/STM32F1/cores/maple/tone.cpp:41:25: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11 [enabled by default]
  HardwareTimer *TTimer[4] { &TTimer1,&TTimer2,&TTimer3,&TTimer4 };
```
It be nice is people who submit new code try it out with the arduino compiler warnings cranked up to the full level and try it for a few different boards before we incorporate the code into the "core".

rogerclarkmelbourne, 2016-12-11:
Hi Rick

OK. Missed that one

The original code only worked for C8 etc (It was supposed to work for "High density" devices but it didnt even compile)

Someone logged a big for the "High density" MCU's which I thought I'd fixed as the compiler was only complaining about where the code tried to init a 4 item array with 8 items, so I fixed that,

But I didn't actually run it on a "High density" board

Can you try replacing the code which defines the number of timers and declares the arrays etc with this

`
#ifdef STM32_HIGH_DENSITY
// define default timer and channel
	#ifndef TONE_TIMER
	#define TONE_TIMER 8
	#endif
	#ifndef TONE_CHANNEL
	#define TONE_CHANNEL 8
	#endif

	HardwareTimer TTimer1(1), TTimer2(2), TTimer3(3), TTimer4(4),TTimer5(5), TTimer6(6), TTimer7(7), TTimer8(8);
	HardwareTimer *TTimer[8] { &TTimer1,&TTimer2,&TTimer3,&TTimer4,&TTimer5,&TTimer6,&TTimer7,&TTimer8 };
#else
	// define default timer and channel
	#ifndef TONE_TIMER
	#define TONE_TIMER 4
	#endif
	#ifndef TONE_CHANNEL
	#define TONE_CHANNEL 4
	#endif

	HardwareTimer TTimer1(1), TTimer2(2), TTimer3(3), TTimer4(4);
	HardwareTimer *TTimer[4] { &TTimer1,&TTimer2,&TTimer3,&TTimer4 };
#endif
`

RickKimball, 2016-12-12:
the problem is that TONE TIMER should be 3

rogerclarkmelbourne, 2016-12-12:
OK

Looks like multiple problems

rogerclarkmelbourne, 2016-12-12:
Rick

I don't think the line 

`HardwareTimer *tone_timer = TTimer[TONE_TIMER]; `

needs to assign 

tone_timer

as it appears to do this again at line 142
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/e71d176618b8b77c727af0891343a278246016fd/STM32F1/cores/maple/tone.cpp#L142

So I've commented out the assignment and it still seems to work ;-)

Also the initialisation of the arrays.

I've changed it to 

HardwareTimer *TTimer[4] =  { &TTimer1,&TTimer2,&TTimer3,&TTimer4 };

and this seems to compile without warnings and as far as I can tell also works fine and the array contains the same values as it did with the other initialisation style
I'm still not entirely sure what the error message about initialisation meant. It sounds like the compiler changed its mode / settings in order to allow that initialisation style and thats what it was complaining about .

But I could be wrong.

BTW. I also fixed 2 other issues. 
tone.h needed to be added to wirish.h and noTone was not working because I changed the argument types to match the SAMD platform, and didnt realise that the original author was calling tone from noTone, and passing negative values

Also, the original code seemed to allow a frequency of 0 as long as the duration was not zero, and vice versa, so I've changed an || to a && as I think it was just a logic mistake and the original author probably intended to test that both frequency and duration were valid (non zero)

RickKimball, 2016-12-21:
Thanks that change seems to take care of it.

Thanks Roger!

closed 2016-12-21 by RickKimball

rogerclarkmelbourne, 2016-12-21:
No worries.

I should have spotted some of the other bugs, (but probably not the initialisation style incompatibility with our default compiler settings ;-) )