rogerclarkmelbourne/Arduino_STM32 #239: STM32F103T building failure

Testato, 2016-12-17:
if you try to compile the simple empty sketch:
```
void setup() {
}

void loop() {
}
```

the building fail:

> D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\Arduino_STM32\STM32F1\variants\generic_stm32f103t\wirish\boards.cpp: In function 'void adc_default_config(const adc_dev*)':
> 
> D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\Arduino_STM32\STM32F1\variants\generic_stm32f103t\wirish\boards.cpp:183:34: error: invalid conversion from 'const adc_dev*' to 'adc_dev*' [-fpermissive]
> 
>      adc_enable_single_swstart(dev);
> 
>                                   ^
> 
> In file included from D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\Arduino_STM32\STM32F1\cores\maple/wirish_types.h:39:0,
> 
>                  from D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\Arduino_STM32\STM32F1\cores\maple/boards.h:38,
> 
>                  from D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\Arduino_STM32\STM32F1\variants\generic_stm32f103t\wirish\boards.cpp:48:
> 
> D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\Arduino_STM32\STM32F1\system/libmaple/include/libmaple/adc.h:306:13: error:   initializing argument 1 of 'void adc_enable_single_swstart(adc_dev*)' [-fpermissive]
> 
>  extern void adc_enable_single_swstart(adc_dev* dev);
> 
>              ^
> 
> D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\Arduino_STM32\STM32F1\variants\generic_stm32f103t\wirish\boards.cpp:184:53: error: invalid conversion from 'const adc_dev*' to 'adc_dev*' [-fpermissive]
> 
>      adc_set_sample_rate(dev, wirish::priv::w_adc_smp);
> 
>                                                      ^
> 
> In file included from D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\Arduino_STM32\STM32F1\cores\maple/wirish_types.h:39:0,
> 
>                  from D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\Arduino_STM32\STM32F1\cores\maple/boards.h:38,
> 
>                  from D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\Arduino_STM32\STM32F1\variants\generic_stm32f103t\wirish\boards.cpp:48:
> 
> D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\Arduino_STM32\STM32F1\system/libmaple/include/libmaple/adc.h:268:6: error:   initializing argument 1 of 'void adc_set_sample_rate(adc_dev*, adc_smp_rate)' [-fpermissive]
> 
>  void adc_set_sample_rate(adc_dev *dev, adc_smp_rate smp_rate);
> 
>       ^
> 
> D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\Arduino_STM32\STM32F1\variants\generic_stm32f103t\wirish\boards.cpp: In function 'void setup_adcs()':
> 
> D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\Arduino_STM32\STM32F1\variants\generic_stm32f103t\wirish\boards.cpp:189:35: error: invalid conversion from 'void (*)(const adc_dev*)' to 'void (*)(adc_dev*)' [-fpermissive]
> 
>      adc_foreach(adc_default_config);
> 
>                                    ^
> 
> In file included from D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\Arduino_STM32\STM32F1\cores\maple/wirish_types.h:39:0,
> 
>                  from D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\Arduino_STM32\STM32F1\cores\maple/boards.h:38,
> 
>                  from D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\Arduino_STM32\STM32F1\variants\generic_stm32f103t\wirish\boards.cpp:48:
> 
> D:\DATI\PERSONALE\Varie\Elettronica\ARDUINO\SKETCH\hardware\Arduino_STM32\STM32F1\system/libmaple/include/libmaple/adc.h:282:13: error:   initializing argument 1 of 'void adc_foreach(void (*)(adc_dev*))' [-fpermissive]
> 
>  extern void adc_foreach(void (*fn)(adc_dev*));
> 
>              ^
> 
> exit status 1
> Error compiling for board Generic STM32F103T series.

rogerclarkmelbourne, 2016-12-18:
Interesting

I'm not sure why the HT-TinyF103T works and the generic version doesnt

I'll need to add it to my to do list to investigate when I get some spare time, perhaps next year.

Testato, 2016-12-20:
it's only 12 days to next year :-)

rogerclarkmelbourne, 2016-12-20:
Depends which calendar you use.

rogerclarkmelbourne, 2016-12-20:
You could of course attempt to look at this yourself.

Testato, 2016-12-21:
I already posted 3 PR for the Ram indication issue, but you do not apply it, so it is unuseful that i solve it myself 😂

stevstrong, 2016-12-21:
Don't expect to happen everything over night. If you want an issue to be solved, provide PRs and be patient.
By providing PR means that you locally already solved your issue, so that you can continue your work with your local version.
By applying PRs Roger will just share your contribution with others.

So my question is: do you know how to resolve the posted error message?

Testato, 2016-12-21:
I posted 3 pr for another error, not this one.
I do not worked on this isdue, only apened it, but is good working on it after the other pr is merged, because the board.txt maybe can interferee.

I have also another PR to propose on a third question, but i want do one by one, so i will wait fir the merge befoure use other spare time to continue my work on this core

florisla, 2017-03-22:
I think I have fixed this compilation issue.  See pull request #259.

florisla, 2017-03-23:
#259 is merged now (thanks for the quick response Roger) so this issue can be closed.

closed 2017-03-23 by rogerclarkmelbourne