rogerclarkmelbourne/Arduino_STM32 #241: 32bit elf in 64bit dist

beginner3456789, 2017-01-06:
I found a 32 bit binary in my 64 bit Linux here:
~/Arduino/hardware/Arduino_STM32/linux64/tools/upload-reset
`upload-reset: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.6.24, BuildID[sha1]=dc8dc5ad6de94179a9d534b1a34aeb6d906640b1, not stripped
`

The 32 bit upload-reset produces an error from the shell. I built a 64 bit version of upload-reset and that now runs without errors except I haven't tested anything.

```
# gcc -o upload-reset upload-reset.c
# ./upload-reset
Usage: upload-reset <serial_device> <Optional_delay_in_milliseconds>

```
Also, I noticed there are three 64bit binaries in the 32 bit dist here:
~/Arduino/hardware/Arduino_STM32/linux/tools/dfu-util/

```
file *

dfu-prefix: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=00bf7e97d0c1f723bf4fd42b15e4e604faed5623, not stripped
dfu-suffix: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=6a2247e79caa6ce6791dd0fef8a63a7e4290bab2, not stripped
dfu-util:   ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.24, BuildID[sha1]=a653b7a3d4a180951bb54db5eeae52111396fc2a, not stripped

```

beginner3456789, 2017-01-21:
My "blue pill" type STM32f103 finally arrived and is working with my 64 bit linux Arduino. I have Void glibc linux64 with Arduino 1.8.0 installed from the distro. Also installed is the distro's openocd software and I installed the generic bootloader using the [ST-Link wiki proceedure](https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Programming-an-STM32F103XXX-with-a-generic-%22ST-Link-V2%22-programmer-from-Linux) with success the first attempt.
There was only really one problem after getting the udev.d/rules working. The Arduino would look for files in the ~/Arduino/hardware/Arduino_STM32/tools/linux/ directory except I removed that because of the 32bit binaries. I had to symlink the ~/Arduino/hardware/Arduino_STM32/tools/linux64/ to the ~/Arduino/hardware/Arduino_STM32/tools/linux/ so the programs would be found. Arduino IDE was initially looking for maple_upload and the dfu-util/ programs when the error was produced. There was no error after the sym-linking.
The "blue pill" then performed normally with the generic bootloader. I could easily upload from the Arduino to the "blue pill" without needing any intervention and the board would reset automatically and immediately run the new program the same as my nano clone.
The [specific "blue pill" board](http://www.ebay.com/itm/141976860562) I am using is similar to the other generic boards except for having a micro-usb socket instead of a mini-usb type. Also, I have not had time to check the usb resistor to see if that is the good 1.5K type or bad 10k type except I see no reason to complain about the function after my first afternoon playing with the board. The simple sketches I have tried have been uploaded and run without intervention and did send data back to the Arduino serial monitor as directed and so I think the 'upload-reset' program is working as designed.
I can see no problems after modifying the paths to force the Arduino to find the correct binaries.

rogerclarkmelbourne, 2017-01-21:
Hi

As these are general notes (apart from the issue with 32/64 bit binaries, you would be better off posting them to the stm32duino community forum

www.stm32duino.com

beginner3456789, 2017-01-21:
I see where this issue is already well known so this item could be closed. ref: http://www.stm32duino.com/posting.php?mode=quote&f=41&p=19108

rogerclarkmelbourne, 2017-01-22:
OK. Closing it.

closed 2017-01-22 by rogerclarkmelbourne