rogerclarkmelbourne/Arduino_STM32 #242: Wrong USB prescaller value for F_CPU = 48 MHz

stevstrong, 2017-01-08:
Hi Roger,
I just saw that the `RCC_PRESCALER_USB`  value [here ](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/variants/generic_stm32f103c/wirish/boards_setup.cpp#L74) should be set to `RCC_USB_SYSCLK_DIV_1`.
Please confirm.
I didn't want to action a PC, I thought you could more easily and quickly change it yourself.
Thanks.

rogerclarkmelbourne, 2017-01-09:
I remember adding in all the code for the GD32, but I think somehow parts of it were lost when I merged the GD32 into the STM32 (as they were virtually identical)

I suspect the GD32 PLL divider was correct, and I had those defines at line 51

I will see if I can find the code in the (now deleted) GD32 sub folder, as git should keep the files even though the are now deleted.

rogerclarkmelbourne, 2017-01-09:
I found the original GD32 code (which was lost when I manually merged)

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/66456b5572f5f4a3b7d43f094b0419227f2d4d2f/GD32F1/variants/generic_gd32f103c/wirish/boards_setup.cpp

I still had the bug with the USB DIV PLL in that code (for 48Mhz) but I set the main PLL as you suggested

rogerclarkmelbourne, 2017-01-09:
Should be fixed by

https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/7dcbbc85b00b23a9ef7ec8692310d90623aeebc1

Note. I noticed that tone() does not give the correct frequency if 48Mhz is selected.

I need to debug and fix this.

stevstrong, 2017-01-09:
My local version for generic C8T8 (BluePill) identical with these changes works fine, so I assume the rest of them should work, too.

rogerclarkmelbourne, 2017-01-09:
Update. 
Also fixed the tone() bug, by using F_CPU in its loops calculation 

see 

https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/c2f4aa8566e6d294e544084f286a5b06f070453e

rogerclarkmelbourne, 2017-01-10:
OK. I'll close this as its fixed

closed 2017-01-10 by rogerclarkmelbourne