rogerclarkmelbourne/Arduino_STM32 #243: Compile Error Arduino 1.6.5

Lexander24, 2017-01-13:
Hello, i want to ask. Why this happen ?

In file included from C:\Users\dell\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/Arduino.h:30:0,

                 from sketch\sketch_jan13a.ino.cpp:1:

C:\Users\dell\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/wirish.h:52:28: fatal error: libmaple/stm32.h: No such file or directory

 #include <libmaple/stm32.h>

                            ^

compilation terminated.

exit status 1
Error compiling for board Maple Mini.
-------------------------------------

Thx

stevstrong, 2017-01-23:
Have you exactly followed the instructions from wiki?
Please try Arduino with at least 1.6.9., better 1.6.12.

rogerclarkmelbourne, 2017-01-23:
This is a support question which should be posted to the forum

www.stm32duino.com

Github is not a place to post questions and support requests, the issue system is designed to be used to post actual bugs.

rogerclarkmelbourne, 2017-05-25:
The Arduino IDE is now at 1.8.x and everything seems to compile fine.

Please use 1.8.2 if you are still having problems

closed 2017-05-25 by rogerclarkmelbourne