rogerclarkmelbourne/Arduino_STM32 #245: Failed to init device

PriankaAg, 2017-01-19:
Hi Roger,
I am using arduino 1.6.5 IDE on ubuntu with CP2103 programmer but get this error 
 
"Failed to init device.
stm32flash Arduino_STM32_0.9

http://github.com/rogerclarkmelbourne/arduino_stm32

Using Parser : Raw BINARY
Interface serial_posix: 230400 8E1

An error occurred while uploading the sketch"

Tx -Rx
Rx - Tx
GND  GND
5v -
RST -RST
boot0 - 1
boot1 -0

Please help me .....

stevstrong, 2017-01-23:
Please post your entire log, including the passed parameters to the flasher like communication port and binary file.

rogerclarkmelbourne, 2017-01-23:
I think these sorts of support questions are best handled on the forum 

www.stm32duino.com as they are rarely a bug or issue in the code.

closed 2018-05-22 by rogerclarkmelbourne