rogerclarkmelbourne/Arduino_STM32 #250: Unable to compile

bkkrunal, 2017-02-09:
I am getting following error msg when I try to compile any sketch. I am using Arduino IDE 1.6.5. I have reinstalled Arduino to C:/Arduino, still not working.
Please help.

> 

Arduino: 1.6.5 (Windows 7), Board: "Generic STM32F103C series, STM32F103C8 (20k RAM. 64k Flash), Serial"

arm-none-eabi-g++: error: C:\Users\KRUNAL~1\AppData\Local\Temp\build5813785701785279306.tmp/core.a: No such file or directory
Error compiling.

>

victorpv, 2017-04-15:
Please post in the forums in stm32duino.com
Issues here are to report issues suspected to be from the code, while the above looks like either some enviroment issue, user error, or missing files.

stevstrong, 2017-04-15:
Currently only Arduino version 1.6.9 and above are supported.

rogerclarkmelbourne, 2017-05-25:
Closed due to inactivity

closed 2017-05-25 by rogerclarkmelbourne