rogerclarkmelbourne/Arduino_STM32 #251: linker error when using usb_midi

chris-zen, 2017-02-12:
I have created a simple sketch to try the new ```usb_midi``` lib being developed at the ```addMidiHID``` branch. I am using PlatformIO, and I have replaced the default framework with a clone for the mentioned branch.

Everything compiles, but it fails when it tries to link everything together:
```
arm-none-eabi-g++ -o .pioenvs/bluepill_f103c8/firmware.elf -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m3 -Wl,-T"jtag_c8.ld" .pioenvs/bluepill_f103c8/src/midict2.ino.o -L/Users/xxx/.platformio/platforms/ststm32/ldscripts -L/Users/xxx/.platformio/packages/framework-arduinoststm32/STM32F1/variants/generic_stm32f103c/ld -L.pioenvs/bluepill_f103c8 -Wl,--start-group .pioenvs/bluepill_f103c8/libFrameworkArduinoVariant.a .pioenvs/bluepill_f103c8/libFrameworkArduino.a -lc -lgcc -lm -lstdc++ -lnosys -lc -Wl,--end-group
.pioenvs/bluepill_f103c8/libFrameworkArduino.a(usb_midi.o): In function `USBMidi':
/Users/xxx/.platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/usb_midi.cpp:76: undefined reference to `vtable for USBMidi'
collect2: error: ld returned 1 exit status
*** [.pioenvs/bluepill_f103c8/firmware.elf] Error 1
```

This is the code:
```c++
#include <usb_midi.h>

#define LED PC13

void setup() {
  MidiUSB.begin();
  pinMode(LED, OUTPUT);
}

void loop() {
  digitalWrite(LED, LOW);
  MidiUSB.sendNoteOn(1, 64, 127);
  delay(2000);
  digitalWrite(LED, HIGH);
  MidiUSB.sendNoteOff(1, 64, 127);
  delay(1000);
}
```

Any idea why is it failing ???

stevstrong, 2017-02-12:
The [GCC FAQ](https://gcc.gnu.org/faq.html#vtables) has an entry on it:

    The solution is to ensure that all virtual methods that are not pure are defined. Note that a destructor must be defined even if it is declared pure-virtual [class.dtor]/7.

chris-zen, 2017-02-12:
@stevstrong is there something I can do without having to modify the arduino code ? like for example sub-classing USBMidi and making sure it has a constructor and destructor ???

Another question, how would the fix for the arduino code look like ?

stevstrong, 2017-02-12:
You have to remove the "virtual" keyword for the functions which are not pure and not defined from [these lines](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/addMidiHID/STM32F1/cores/maple/usb_midi.h#L197).

chris-zen, 2017-02-12:
@stevstrong that worked, thank you very much !

I guess this is a bug in usb_midi ? if not, then a problem with the compiler flags that PlatformIO is using by default ?

stevstrong, 2017-02-12:
I don't know if it is a bug of the compiler or not, but obviously there are not too many people using it.

I had recently exactly the same issue with another lib which I am porting it to STM32 with the "official" Arduino_STM32 framework, that's why I knew the solution :)

I suggest you to send a PR for Roger with the solution if it works.

sweetlilmre, 2017-02-12:
Wouldn't subclassing the midi class and implementing stubs for the virtual
functions work? That seems to be the design intent.

chris-zen, 2017-02-12:
@sweetlilmre it seems to be two ways to use it by design, one is inheriting as you mention, in which case the user is forced to implement all of the handlers. But the other way is by using the already instantiated MidiUSB (see [here](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/addMidiHID/STM32F1/cores/maple/usb_midi.h#L216))

It looks like there was a solution already in place, but for some reason it was commented out (see [here](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/addMidiHID/STM32F1/cores/maple/usb_midi.cpp#L508-L529)). My solution was just to uncomment it (and remove the example digitalWrites) so there is a default implementation for the handlers.

sweetlilmre, 2017-02-12:
+1 Seems like the correct way to go.

<https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/addMidiHID/STM32F1/cores/maple/usb_midi.cpp#L508-L529>).

feurig, 2017-02-13:
I am amazed to see the old usb maple midi stuff still out there. I wrote it at a time when the arduino midi library was GPL3 so it pretty much had to be scratch written if you wanted to incorporate it into a commercial project. Since then the arduino midi lib has adapted a more flexible license and more of their code can be leveraged.

Has anyone looked at porting the pluggable architecture that the arduino team is currently using?
It makes HID and MIDI pretty sane in a hardware independent context.

For example this is a library for midi usb.

https://www.arduino.cc/en/Reference/MIDIUSB

I have been working with the samd lately but want to get back over here to the STM and would probably start porting the current pluggable based usb midi implementation, please feel free to email me if any of you are interested in persuing this approach as well.

Don.

rogerclarkmelbourne, 2017-02-13:
Thanks Don

I've not seen the pluggable architecture that Arduino are using.

Currently things are in a state of flux with the STM cores, as we have the old LibMaple based core (this one) and a new core being written my STM themselves (see github.com/stm32duino)

I initially thought that the STM core would eventually replace the LibMaple based core and we could use their USB code and use a composite USB device. But things arent that simple, because STM's core uses the HAL which causes a load of bloat, and also the STM core was initially written for the STM32L4 which has a lot of RAM and Flash and was not optimised for the F103 or the even smaller processors.

So because the Midi / HID etc is only used by a small proportion of the Arduino for STM32 users, I thought it was safer to have a branch of the repo with this in it, as it would save a lot of confusion (mainly because you have to pick the USB support you want from a menu in the IDE and can't just get it by calling a function in the code)

If you are thinking about getting back into STM32 dev for Arduino, check out our forum on www.stm32duino.com

rogerclarkmelbourne, 2018-05-22:
USB handling has been reworked by @arpruss 
Please retest with the latest version using the lib @arpruss  wrote, and re-open if this is still an issue

closed 2018-05-22 by rogerclarkmelbourne