rogerclarkmelbourne/Arduino_STM32 #253: SPI compatibility issue?

tcurdt, 2017-03-12:
I was trying to build the minimal U8G2 example (using platformio) for arduino on a STM32F103C8 (and Arduino Uno as reference)

```
brew install platformio
mkdir test && cd test
platformio init --board genericSTM32F103C8 --board uno
platformio lib install U8g2
```

Change the `platformio.ini` to

```
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
lib_deps = U8g2

[env:uno]
platform = atmelavr
board = uno
framework = arduino
lib_deps = U8g2
```

Add a `main.cpp`

```
#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>
#include <Wire.h>

U8G2_SSD1306_128X64_NONAME_1_4W_SW_SPI u8g2(U8G2_R0, 13, 11, 10, 8);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0,24,"Hello World!");
  } while ( u8g2.nextPage() );
}
```

When building I am getting:

```
$ platformio run
[Sun Mar 12 21:41:56 2017] Processing genericSTM32F103C8 (platform: ststm32, lib_deps: , board: genericSTM32F103C8, framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 25 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <U8g2> v2.14.7
|   |-- <SPI> v1.0
|   |-- <Wire>
|-- <SPI> v1.0
|-- <Wire>
Compiling .pioenvs/genericSTM32F103C8/lib/U8g2_ID942/U8x8lib.o
Compiling .pioenvs/genericSTM32F103C8/lib/U8g2_ID942/clib/u8g2_cleardisplay.o
Compiling .pioenvs/genericSTM32F103C8/lib/U8g2_ID942/clib/u8g2_d_memory.o
Compiling .pioenvs/genericSTM32F103C8/lib/U8g2_ID942/clib/u8g2_d_setup.o
Compiling .pioenvs/genericSTM32F103C8/lib/U8g2_ID942/clib/u8g2_font.o
.piolibdeps/U8g2_ID942/src/U8x8lib.cpp: In function 'uint8_t u8x8_byte_arduino_hw_i2c(u8x8_t*, uint8_t, uint8_t, void*)':
.piolibdeps/U8g2_ID942/src/U8x8lib.cpp:469:7: error: 'class TwoWire' has no member named 'setClock'
Wire.setClock(400000L);
^
*** [.pioenvs/genericSTM32F103C8/lib/U8g2_ID942/U8x8lib.o] Error 1
=============================================================== [ERROR] Took 1.99 seconds ===============================================================

[Sun Mar 12 21:41:58 2017] Processing uno (platform: atmelavr, lib_deps: U8g2, board: uno, framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 29 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <U8g2> v2.14.7
|   |-- <Wire> v1.0
|   |-- <SPI> v1.0
|-- <Wire> v1.0
|-- <SPI> v1.0
Calculating size .pioenvs/uno/firmware.elf
AVR Memory Usage
----------------
Device: atmega328p

Program:    9060 bytes (27.6% Full)
(.text + .data + .bootloader)

Data:        568 bytes (27.7% Full)
(.data + .bss + .noinit)

============================================================== [SUCCESS] Took 1.13 seconds ==============================================================

======================================================================= [SUMMARY] =======================================================================
Environment genericSTM32F103C8	[ERROR]
Environment uno               	[SUCCESS]
=============================================================== [ERROR] Took 3.12 seconds ===============================================================
```

Any thoughts?

rogerclarkmelbourne, 2017-03-13:
Please register on www.stm32duino.com and post the question there, so that the community can give you some feedback

closed 2017-03-13 by rogerclarkmelbourne

tcurdt, 2017-03-13:
Yay - a forum :-(