rogerclarkmelbourne/Arduino_STM32 #254: EEPROM ISSUE

cheeta1, 2017-03-14:
hi rogerclarkmelbourne,

the EEPROM have some issues. whenever i reset the stm32f103C8 the EEPROM is erased completely and every block is set to 0xFFFF (which is the default value that you set in EEPROM Library); i read that you mentioned the EEPROM might be erased during programming the controller but it is also being erased after reset. can you suggest me what are the causes for that and where to look which files and which functions should i change to resolve this issue?

rogerclarkmelbourne, 2017-03-14:
This isnt normal

I have used the EEPROM library and it worked fine when I uploaded via the bootloader

Its also never erased after rebooting

No one else has reported this problem and I know many people use this feature

How are you uploading?

Can you demonstrate the problem using the example sketches in the library ?

cheeta1, 2017-03-14:
i am using STLink V2 programmer to upload the code. i tried the example of eeprom. when i reset the controller and read the eeprom it send the status of (0x82) which is bad address in the header file but when i write the same adress and read it again it gives me the correct value and when i reset it again it show bad address on reading the eeprom.

cheeta1, 2017-03-14:
EEPROM.PageBase0 = 0x801F000;
EEPROM.PageBase1 = 0x801F800;
EEPROM.PageSize  = 0x400;

i am using stm32f103c8  so i put these lines in the code to define eeprom address and size but some how these line are causing the problem. i formatted the eeprom using EEPROM.format(); and removed the above mentioned lines from code and it seemed to be working correctly.

can you tell me the reason for that.

thanks

rogerclarkmelbourne, 2017-03-16:
I didnt write that library. I think you will need to either contact the original author, if you can track them down, or you will need to debug the library functionality.

closed 2017-04-14 by cheeta1