rogerclarkmelbourne/Arduino_STM32 #255: Serial monitor problem

Mario857, 2017-03-16:
Serial monitor does not work properly on Linux,
Sometimes I can see the readings sometimes not... Window allways opens...
and I had problems on Windows,but not in the Arduino IDE instead in my c# app every time I read from port I received "1" from serialport function or It couldnt connect,same thing happened in java, everything works fine if use arduino nano or any other...

rogerclarkmelbourne, 2017-03-16:
On the whole both linux and Windows serial works OK.

Linux needs changes to the UDEV rules, but I presume you did that.

Someone else reported an issue on Windows a few days ago, using C in Visual Studio, and the problem was that the core needs DTR to be set before it will send data.

The Arduino IDE, and some other terminals, set DTR when they are opened, but not all terminals do this

As a temporary fix, until I know how stable the fix is, I have added a new branch 
https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/ZUMspot

Which does not require DTR to be set by the host, and this fixed the problem on Windows.

So I'd recommend you try that branch and see if it fixes your problem

Ideally I'd like to put that fix into the Master branch, but it will need more testing by members of the Arduino STM32 community before I can do that.

BTW, ZUMspot is a ham radio repeater system, and the fix was for them.
Though they only had a problem on Windows, and it worked fine on Linux

tcurdt, 2017-03-25:
I am uploading the build through a stlinkv2 (SWD). When I try open the console in the Arduino IDE it says `/dev/cu.wchusbserial410 is not available`

Would I need to connect the board's (STM32F103C8) usb port as well for this to work? or how is serial monitor support supposed to work for the stm32 platform?

I am on macOS 10.12.3 using Arduino IDE 1.8.1

rogerclarkmelbourne, 2017-05-25:
Try the latest version, it has a fix that delays the upload script returning until the serial device has been enumerated. 
Please re-open this if the problem persists but I'm closing now as there has been no activity on this issue for 2 months

closed 2017-05-25 by rogerclarkmelbourne