rogerclarkmelbourne/Arduino_STM32 #256: macOS: libusb is not included, that dfu-util depends on

toyoshim, 2017-03-16:
https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation

The document says now that dfu-util is included, users do not need to install it manually.
Actually, a binary is included, but it seems users still need to install libusb on which the distributed binary depends. Otherwise, it just aborts due to missing dylib on starting upload.

rogerclarkmelbourne, 2017-03-16:
What would you recommend to fix this ?

toyoshim, 2017-03-17:
Is it easy to build another dfu-util with a static library version of libusb and to push it into the repo.?

rogerclarkmelbourne, 2017-03-17:
I don't know

You could give it a try and make a PR with the updated version

satoshinm, 2017-12-04:
I had the same problem, not sure of a proper fix but this seems to work as a temporary workaround:

```sh
brew install libusb
sudo mkdir -p /opt/local/lib
sudo ln -s /usr/local/Cellar/libusb/1.0.21/lib/libusb-1.0.0.dylib /opt/local/lib/libusb-1.0.0.dylib
```

rogerclarkmelbourne, 2017-12-04:
@satoshinm 

Just out of curiosity, which version of MacOS are you using ?

satoshinm, 2017-12-04:
Just updated to the latest, macOS High Sierra 10.13.2 public beta 6 (but not sure if using the beta is significant)

JBerg60, 2018-02-10:
Same error here. Solution was to install libusb `brew install libusb`

Cipher2600, 2018-05-27:
I am getting the following error upon attempting to compile:

fork/exec /bin/arm-none-eabi-g++: no such file or directory
Error compiling for board Generic STM32F103C series.

I have done everything as stated and also built the libusb. I am new to this so any help would be great! I am running Sierra on a MacBook Pro.

ratkins, 2019-02-28:
Same issue for me on Mojave (10.14.3 (18D109)). `brew install libusb` (no other `ln` commands necessary) fixes it.

stevstrong, 2019-09-16:
Is this solved? Possible duplicate of #328 
Anyway, I have no Mac, so any help on this is appreciated.

rogerclarkmelbourne, 2020-05-18:
This issue has no activity for a long time , so I will close it

closed 2020-05-18 by rogerclarkmelbourne