rogerclarkmelbourne/Arduino_STM32 #257: ST-link upload problem on Linux

Mario857, 2017-03-17:
After reinstalling Ubuntu 16.04 I have some problems...
I've installed Arduino IDE run it as a root, set udev rules...
Code compiles but wont upload...
Any idea how to fix the problem?

/home/mario/Documents/arduino-1.6.12/hardware/Arduino_STM32-master/tools/linux/stlink_upload COM1 {upload.altID} {upload.usbID} /tmp/arduino_build_593833/sketch_mar17a.ino.bin 

/home/mario/Documents/arduino-1.6.12/hardware/Arduino_STM32-master/tools/linux/stlink_upload: line 2: /home/mario/Documents/arduino-1.6.12/hardware/Arduino_STM32-master/tools/linux/stlink/st-flash: No such file or directory

Mario857, 2017-03-19:
Problem was fixed by installing  :
1. http://apt.ubuntu.com/p/libc6-i386
2. apt-get install libusb-1.0-0:i386

closed 2017-03-19 by Mario857

smv1210, 2017-12-22:
I suppose, that you do use 64bit ubuntu version.

If it is so, then it would be better in .../Arduino_STM32-master/tools/ to :
- delete "linux" and 
- rename "linux64" to "linux" sub-folders
 rather than installing those :i386 packages