rogerclarkmelbourne/Arduino_STM32 #265: Possible typo or change in libmaple? (STM32F1 -> STM32ADC -> util_adc.h)

SuitableOrNot, 2017-04-09:
I was unable to compile the ADC-Examples provided with Arduino_STM32 for my "BluePill"-Board (STM32F103CB).
The compiler mourned about "ADC_CR2_TSEREFE" not being defined in `STM32F1/libraries/STM32ADC/src/utility/util_adc.h` on line 48:
`void enable_internal_reading(adc_dev *dev) {`
`   dev->regs->CR2 |= ADC_CR2_TSEREFE;`
`}`

After following the references (via Eclipse/Sloeber) I found the list of defines in `libmaple/stm32f1/include/series/adc.h` to only have:
`#define ADC_CR2_TSVREFE_BIT             23`
`// [...]`
`#define ADC_CR2_TSVREFE                 (1U << ADC_CR2_TSVREFE_BIT)`

So I went ahead and changed the above line to:
`void enable_internal_reading(adc_dev *dev) {`
`   dev->regs->CR2 |= ADC_CR2_TSVREFE;`
`}`

It compiled fine and also seems to sample correctly, but as this is my first stm32-project,
I was not 100% sure of it's correctness, so I searched through STM's Reference Manual "RM0008" and found the ADC register map on page 252 (of 1137),
which also described bit 23 on Offset "ADC_CR2" to be "TSVREFE".

stevstrong, 2017-04-09:
The correct version is ADC_CR2_TSVREFE.
This was corrected [in the repo 9 month ago](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/754bc2969921f1ef262bd69e7faca80b19db7524/STM32F1/system/libmaple/stm32f1/include/series/adc.h#L111). It may happen that the ADC lib is older so still working with the old definition. It should be corrected as well.

rogerclarkmelbourne, 2017-05-25:
@stevstrong 
I have attempted to correct the typo in the F4 repo that you noticed.

See 

https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/d3058d39f077daceea2573fd1fbe6255b5cc1497

F4 seems to still compile OK, so I think my change is correct

closed 2017-05-25 by rogerclarkmelbourne