rogerclarkmelbourne/Arduino_STM32 #266: Cannot upload to board

klyman, 2017-04-13:
I have issue with uploading sketches to my STM32F103C8T6 based board.

First of all, I have installed generic bootloader and checked with Arduino IDE - and THIS IS WORKING. I am able to upload sketches successfully.

An issue occurs when I try to do the same in Sloeber. All of configuration seems to be fine. I am able to compile sketches, but uploading gives me following output in the console:

```
Starting upload
using arduino loader
Starting reset using DTR toggle process
Toggling DTR
Continuing to use COM5
Ending reset

Launching 
Output:
failed to upload
```

It looks like Sloeber cannot extract correct upload tool. Any ideas why is it happening or how to fix it?

I am using:
Arduino_STM32 https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/e3753df280123427a07d74f5b142d927a5546cf7
Arduino IDE 1.8.2
Sloeber 4.0.0

klyman, 2017-04-13:
Basing on environment variables which are setup in project I have successfully extracted command which should be called and executed it from command line. It worked.

```
D:\development\tools\Arduino_STM32\tools\win>maple_upload.bat COM5 2 1EAF:0003 D:\development\workspace\distance\distance.bin
maple_loader v0.1
Resetting to bootloader via DTR pulse
Searching for DFU device [1EAF:0003]...
Found it!

Opening USB Device 0x1eaf:0x0003...
Found Runtime: [0x1eaf:0x0003] devnum=1, cfg=0, intf=0, alt=2, name="STM32duino bootloader v1.0  Upload to Flash 0x8002000"
Setting Configuration 1...
Claiming USB DFU Interface...
Setting Alternate Setting ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
Transfer Size = 0x0400
bytes_per_hash=320
Starting download: [##################################################] finished!
```

klyman, 2017-04-13:
Not sure how exactly it works, but there is no `A.JANTJE.UPLOAD.TOOL` variable defined in project and defining it manually does not work

SalvorinFex, 2017-04-13:
@klyman Well, in my knowledge of just being another STM32 user, this repo only supports the original Arduino IDE by arduino.cc, so not sure how slober fits into that.
It might not be compatible with this library.

stevstrong, 2017-04-13:
In addition, such kind of issues should be normally first discussed in forum: www.stm32duino.com.
Please register and post the issue there, even if it obviously not an STM32duino related one.

victorpv, 2017-05-09:
Just to add that I have seen similar issues myself with sloeber not pulling the right variables. Normally get resolved by selecting a different board, and then the one you are actually want.
If I remember right, in the last stable sloeber I had to select a Due board, then a maple mini, then I can select a generic board.
Doesn't happen if you just clone a project that already has the right board selected, so I tend to clone and replace the ino files.
The last nightly build has some bug fixes, some may be related to this (not certain of that).
I haven't investigated to find if there is something in the boards.txt file causing this, if you do and find the cause, please post on the forum as Steve advised.

rogerclarkmelbourne, 2017-05-25:
Closed due to inactivity

closed 2017-05-25 by rogerclarkmelbourne