rogerclarkmelbourne/Arduino_STM32 #268: how to upload stlink for f4

sonsatthu507, 2017-04-14:
Hi stevstrong, how to upload stlink for f4 and i build error STM32F4\cores\maple/libmaple/gpio.h:35:20: fatal error: gpioF4.h: No such file or directory
 #include "gpioF4.h"

stevstrong, 2017-04-14:
It looks like you are using my latest changes regarding F4.
I think you are in the wrong place.
First, my F4 changes were not yet committed in this repo. You may try [my repo](https://github.com/stevstrong/Arduino_STM32/tree/Black_F4_variant) instead.
Second, such kind of questions are better placed in the forum: www.stm32duino.com.
Please register and post your issue there.

sonsatthu507, 2017-04-14:
Thank Stevstrong!

stevstrong, 2017-04-14:
Please close the issue, if you think is appropriate.

closed 2017-04-14 by sonsatthu507