rogerclarkmelbourne/Arduino_STM32 #269: update to arduino 1.8

truebucha, 2017-04-18:
I use arduino 1.8 to flash my controllers so I move a part of you repo to package STM32f. 
I planning to place it to github or possible you interested in move you project to this modern structure?

rogerclarkmelbourne, 2017-05-11:
I'll need to check what changes there were in version 1.8

However backwards compatibility would need to be maintained for a long time, so I could not make changes that only work with V1.8 at the moment

Also BTW. I presume you package file is a demo / placeholder as some of the zip files referenced in it. do not exist
e.g. 
https://github.com/2.3.0.zip

truebucha, 2017-05-11:
you are right. this is a mac version of you library, tat was made just to work. the right implementation should store in the wed package, tools, and the json separately bundled in zip. I just recreate extracted folders structure and fix some configuration files to work. It could be more easy for you to check my changes like example rather implemented all from scratch.

closed 2017-05-11 by truebucha