rogerclarkmelbourne/Arduino_STM32 #284: Serial.flush() should not empty transmit buffer

FabianInostroza, 2017-05-12:
Currently the call to Serial.flush() empties the transmit buffer, however the Arduino documentation says this behavior changed after Arduino 1.0 [1].

This causes problems with lib TinyGSM.

[1] https://www.arduino.cc/en/Serial/Flush

FabianInostroza, 2017-05-12:
This commit should be reverted
https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/7ba90658950ea530526e6f6f8a6aa452b804fefa

rogerclarkmelbourne, 2017-05-13:
Thanks.

@edogaldo submitted that update, I wonder if he can comment on this

edogaldo, 2017-05-13:
Hi, I see the point to change flush behaviour to make it consistent with
Arduino 1.0 api but can't see the point to just revert that commit..

rogerclarkmelbourne, 2017-05-13:
Thanks @edogaldo 

Perhaps @FabianInostroza  can suggest a fix, i.e add a while loop which 

"Waits for the transmission of outgoing serial data to complete" (from Arduino API page)

FabianInostroza, 2017-05-13:
Replacing
`usart_reset_tx(this->usart_device);`
with
`while( !rb_is_empty(this->usart_device->wb) );`
in `HardwareSerial::flush(void)` works but I haven't done extensive testing and I think it would be better to add a new function to usart.h

rogerclarkmelbourne, 2017-05-13:
I'm not sure why it would be better in usart.h, its still not going to be interrupt blocking??

FabianInostroza, 2017-05-13:
My comment about a new function in usart.h was about code organization, not about blocking the interrupt.

This should be safe, I'm not sure if disabling the interrupt is needed. Reading the head index of the ring buffer should be an atomic operation on an ARM and the tail index of the buffer should not be modified inside an interrupt, so it _maybe_ not necessary.
`
void HardwareSerial::flush(void) {
    usart_reset_rx(this->usart_device);
    while( 1 ){
	this->usart_device->regs->CR1 &= ~((uint32)USART_CR1_TXEIE);
	if( rb_is_empty(this->usart_device->wb) ){
	    break; // if buffer is empty the interrupt should be disabled, so we left it disabled...
	}
	this->usart_device->regs->CR1 |= ((uint32)USART_CR1_TXEIE);
    }
}
`

rogerclarkmelbourne, 2017-05-14:
@FabianInostroza 

OK. I'll think about the code organisation, as I can't see any compelling argument to having the code in usart.h as opposed to hardwareserial

If it needs to be called from elsewhere, I suppose usart.h would be the best place for it, but I don't know if that's the case

Also, I suppose we also need to consider the USB serial functionality and where the same code would need to be added to that, so that it conforms with the Arduino API

edogaldo, 2017-05-14:
Just a tip: I checked the DUE implementation and it's pretty simple, I'd stick with a simple implementation too

rogerclarkmelbourne, 2017-05-14:
@edogaldo 

Thanks. 

My motto is KISS, so I'm happy to keep this as simple as possible and just change hardwareserial.cpp

edogaldo, 2017-05-15:
Hi Roger, FabianInostroza, I just pushed PR #287 to address this issue.
The solution is based on the DUE implementation.
Tests performed on a blue pill seem ok.

Best regards, E.

rogerclarkmelbourne, 2017-05-25:
Fixed in

https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/8c433e3c7186877f65acf91f975300d3d0825111

Note. because I had not pulled before committing, it eventually also got committed in 

https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/ccddc1f354d2b645da9615057324be744a1be2d6

closed 2017-05-25 by rogerclarkmelbourne