rogerclarkmelbourne/Arduino_STM32 #289: [STM32F1] endTransmission not sending stop condition / not accepting boolean

matztam, 2017-05-20:
> As of Arduino 1.0.1, endTransmission() accepts a boolean argument changing its behavior for compatibility with certain I2C devices. 
https://www.arduino.cc/en/Reference/WireEndTransmission

The wire-lib for STM32F1 only implements endTransmission(void) and seems not to send stop condition.Or maybe i'm just not getting it ;)

rogerclarkmelbourne, 2017-05-22:
Thanks.

It looks like endTransmission(boolean stop)  is a feature which was not originally implemented when the library was written for Arduino version 0022, and no one has noticed until now that Arduino 1.0.1 added this additional feature.

Can you suggest a solution to this?

The Wire library has this command

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/4021c32fdb809f16a61c073d04e1d8c2c487a767/STM32F1/libraries/Wire/Wire.cpp#L73-L77

but its only implemented for software I2C

It could be called from endTransmission , but I am not an expert in I2C, so I don't know to call it in

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/4021c32fdb809f16a61c073d04e1d8c2c487a767/STM32F1/libraries/Wire/WireBase.cpp#L62-L71

e.g. at the start of the function , just before

 if (tx_buf_overflow) {

???

rogerclarkmelbourne, 2017-07-31:
See the other issue, which is now closed

closed 2017-07-31 by rogerclarkmelbourne