rogerclarkmelbourne/Arduino_STM32 #295: Wire library doesn't support Wire.endTransmission(false); problem in using adafruit mpr121 library.

engrziarehman, 2017-05-26:
I have successfully used MPR121 touch sensor with arduino mega board. but facing problem in compiling with stm32f103c8 board, since the wire library doesn't support repeated start feature. following is the code that is unable to compile
```

uint8_t Adafruit_MPR121::readRegister8(uint8_t reg) {
    Wire.beginTransmission(_i2caddr);
    Wire.write(reg);
    **Wire.endTransmission(false);**
    while (Wire.requestFrom(_i2caddr, 1) != 1);
    return ( Wire.read());
}

uint16_t Adafruit_MPR121::readRegister16(uint8_t reg) {
    Wire.beginTransmission(_i2caddr);
    Wire.write(reg);
    **Wire.endTransmission(false);**
    while (Wire.requestFrom(_i2caddr, 2) != 2);
    uint16_t v = Wire.read();
    v |=  ((uint16_t) Wire.read()) << 8;
    return v;
}
```

rogerclarkmelbourne, 2017-05-26:
Isnt this a duplicate of https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/289 ?

rogerclarkmelbourne, 2017-05-26:
Just a thought....
But if you have hardware which requires this, perhaps you can test the change I proposed in #289

engrziarehman, 2017-07-14:
the change you proposed in #289 is for adding a stop condition, but i have a problem to not to send stop condition during Wire.endtransmission(); because MPR121 is the device that works on repeated start method, means starting transmission again without sending stop. so Arduino Wire library has this feature Wire.endtranmission(false); can be used for this purpose, as it is used in adafruit library. but Arduino_STM32 Wire library doesn't take this boolean argument. what change will i be required?

rogerclarkmelbourne, 2017-07-16:
OK. 

I'll take a look.

rogerclarkmelbourne, 2017-07-31:
This was fixed a while ago. Please retest and reopen if the problem presists

closed 2017-07-31 by rogerclarkmelbourne