rogerclarkmelbourne/Arduino_STM32 #296: Abnormal flash space consumption for trivial code

RickKimball, 2017-05-28:
A user posted a problem with some trivial code over on stm32duino.com

I'm very new to STM32 Arduino so pardon me if I did something silly there, but I don't understand what's going on. When compiled for the STM32F3 Discovery board, the following code occupies 71960 bytes (66%) of program storage space. 7248 bytes of dynamic memory seems too much, too (versus 4008 for an empty .ino).
```
CODE: SELECT ALL
class A 
{
public:
  A() {}

  static A& inst()
  {
    static A a;
    return a;
  }
};

void setup() {
  A::inst();  
}

void loop() {}
```
Note that this is the minimal example. Removing the constructor of A (using the default compiler-generated one) removes the issue. Either a constructor or destructor is required, doesn't matter which. The singleton part is also required, that's what somehow triggers the issue.

RickKimball, 2017-05-28:
The solution is to add the -fno-threadsafe-statics to the cpp.flags:
```
$ git diff -w platform.txt
diff --git a/STM32F1/platform.txt b/STM32F1/platform.txt
index 3da3070..f12ffa1 100644
--- a/STM32F1/platform.txt
+++ b/STM32F1/platform.txt
@@ -22,7 +22,7 @@ compiler.c.elf.flags=-Os -Wl,--gc-sections
 compiler.S.cmd=arm-none-eabi-gcc
 compiler.S.flags=-c -g -x assembler-with-cpp -MMD
 compiler.cpp.cmd=arm-none-eabi-g++
-compiler.cpp.flags=-c -g -Os {compiler.warning_flags} -std=gnu++11 -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_{build.variant} -D{build.vect} -DERROR_LED_PORT={build.error_led_port} -DERROR_LED_PIN={build.error_led_pin} 
+compiler.cpp.flags=-c -g -Os {compiler.warning_flags} -std=gnu++11 -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_{build.variant} -D{build.vect} -DERROR_LED_PORT={build.error_led_port} -DERROR_LED_PIN={build.error_led_pin} -fno-threadsafe-statics
 compiler.ar.cmd=arm-none-eabi-ar
 compiler.ar.flags=rcs
 compiler.objcopy.cmd=arm-none-eabi-objcopy
```

rogerclarkmelbourne, 2017-05-29:
OK

I'm willing to give it a go,

Does anyone else have any thoughts on this

stevstrong, 2017-05-29:
`-fno-threadsafe-statics` is safe.

stevstrong, 2017-07-04:
Oh, and I think to use the A construct within the A class is not quite "trivial"...
But the solution does not harm.

victorpv, 2017-07-04:
The solution could harm if using an RTOS, and there are multiple people using RTOS.
"You can use this option to reduce code size slightly in code that doesn’t need to be thread-safe."

I would not add it without testing extensively with RTOS sketches. Or make it an option in boarts.txt

But given that it solves a very specific issue, and can cause others, personally I think we should skip this "fix"

stevstrong, 2017-07-04:
Sorry, I didn't think on RTOS because I'm not using it :)
Anyway, I tried it an for non-OS solution works fine.

rogerclarkmelbourne, 2017-07-31:
OK. So we have to live with the size as we support RTOS.

closed 2017-07-31 by rogerclarkmelbourne

RickKimball, 2017-08-01:
:( and this is why we can't have nice things

rogerclarkmelbourne, 2017-08-01:
I can't have nice things because I take them apart and sometimes break them ;-)

RickKimball, 2017-08-02:
Fwiw the forum thread shows a work around without the flag

rogerclarkmelbourne, 2017-08-02:
OK.

Thanks

kolyshkin, 2018-02-16:
@victorpv Thanks for you input on the matter. Please let me shed some more light.

> The solution could harm if using an RTOS, and there are multiple people using RTOS.
> "You can use this option to reduce code size slightly in code that doesn’t need to be thread-safe."

The "slightly" here applies to the code which already uses libstdc++, as the savings come from missing calls to to `__cxa_guard_acquire()` and `__cxa_guard_release()` around static local initialization.

For the case described in https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/444, this reduces the binary size from 73K down to 14K (and the RAM used from 6K to 3K), which is not just "slightly", but also makes the whole thing work, as the hardware features 64K flash.

> I would not add it without testing extensively with RTOS sketches.

I am not very familiar with RTOS and the hardware it supports, perhaps you can answer a question -- are there any multicore STM32 CPUs? I goggled around a bit and it looks like it's not the case.

Note that the functionality that is being disabled only helps in case of real multicore CPUs and multithreaded software (i.e. doing some software-only multithreading is fine).

> But given that it solves a very specific issue, and can cause others, personally I think we should skip this "fix".

Again, this fix makes it possible to make an Arduino sketch fit the target hardware flash. Indeed this is a very specific issue, but a very critical one.

Please reconsider. Also, I'll be happy to do some testing with RTOS, in case multicore hardware exist -- if that's the case, I'll be obliged if you can provide some pointers.

riodda, 2018-02-16:
I think that the percentage of people using FreeRTOS and STM32duino port is limited, i think that most of the users will benefit the memory savings.

stevstrong, 2018-02-16:
And all this because a programmer wants to declare static variable within a function...
Wouldn't be easier to change that?

rogerclarkmelbourne, 2018-02-16:
Actually quite a lot of people seem to use FreeRTOS

victorpv, 2018-02-16:
@kolyshkin thanks for the comments and the extra information.
A couple of clarifications first:
I was quoting from the GCC documentation, "slightly" was not a word I chose, but a word that comes from the documentation on that option. And they probably use slightly for a reason, but is not up to me to explain why, the key for me to quote it was about the thread safe part.

About multicore, there is no STM32 multicore that I know of, and anyway FreeRTOS doesn't support multicore (only third party modifications do, not our case), but in the option I did not see any mention to multicore, only to threads. Threads do not need different cores to run, they can run in the same core with time slicing. Could you add some references that explain this only affects in case of multicore CPUs? that will shed some light on when it will pose a problem and when it will not.

kolyshkin, 2018-02-18:
> And all this because a programmer wants to declare static variable within a function...
> Wouldn't be easier to change that?

Having a static within a function is perfectly normal C/C++ code, and changing it would most probably make the code worse / less readable. I agree it could a workaround for a simple sketch, but in my case it's a whole library (FastLED) which may use static locals in a dozen places.

RickKimball, 2018-02-18:
@kolyshkin I found it is easier to just make this project work for yourself rather than trying to meet all the different needs of everyone else.  You might be able to make your changes using the file platform.local.txt. Me? I've just replaced platform.txt with my own.  I also have a makefile that appends the boards.txt with my own custom entries.  https://gist.github.com/RickKimball/71a388efd58ac0d3a0e1a272ecf4e095

kolyshkin, 2018-02-18:
> And they probably use slightly for a reason, but is not up to me to explain why, the key for me to quote it was about the thread safe part.

> they can run in the same core with time slicing.

From my understanding, whether the lack of guarding for static local initialization might lead to runtime bugs depends on what thread switching model (ie scheduler) is used (is it preemptive or voluntary). I now think I was wrong that only multicore/SMP systems are affected. I need to look deeper into how thread scheduling/switching works in RTOS. I found some explanation here: https://www.freertos.org/implementation/a00006.html (also see other articles in https://www.freertos.org/implementation/a00002.html). So, from the first glance it looks like preemptive sheduling is used, i.e. a task can be preempted/interrupted at any time and any other task (including another thread of the same program) can be run.

One other thing to figure out is whether any code compiled with the flag we discuss is running multiple tasks.

What the aforementioned gcc mechanism does is protects the two (or more) threads sharing the same memory from running a static local variable initialization more than once. Here is an example, from the top of my head:

```c
int f(void) {
   // do something
   return ret; // some value
}

int func(void) {
   static int x = f();
}
```

Now, C/C++ standard guarantees that `f()` is only called once, when `func()` is called for the first time. In case a program is multithreaded (comprises of many tasks sharing memory and code, let's assume 2, calling them task1 and task2), the following may happen:

1. `task1` calls `func()`.
2. `func()` calls `f()`.
3. RTOS kicks in, suspends `task1` and runs `task2`.
4. `task2` calls `func()`.
5. `func()` calls `f()` (as x is still not initialized).

Here is the bug -- `f()` is called twice which is a violation of C/C++ standard. This is why GCC wraps the call to `f()` with `__cxa_guard_acquire()` and `__cxa_guard_release()` -- to make sure it won't be called twice.

So, the question remains -- is any code which compiles with options from platform.txt/boards.txt actually runs more than a single task? My current (limited) understanding is it's not the case.

kolyshkin, 2018-02-18:
> So, the question remains -- is any code which compiles with options from platform.txt/boards.txt actually runs more than a single task? My current (limited) understanding is it's not the case.

If anyone more familiar with the matter (e.g. RTOS) can shed some light on this matter, I'll be much obliged.

kolyshkin, 2018-02-19:
> is any code which compiles with options from platform.txt/boards.txt actually runs more than a single task?

OK, I did some more research; looks like yes, this is the case so `-f-no-threadsafe-statics` might be harmful when used with RTOS. From the other side, if this flag is not present, some sketches generated might not fit the RAM of a Maple Mini or Blue Pill board. Both outcomes are bad.

So, there is a need to have a way to add this flag conditionally. Ideally, add it unless RTOS is being used. For the moment I have no idea how to implement it.

kolyshkin, 2018-02-19:
One way would be to add something like `cppflags=-fthreadsafe-statics` to `FreeRTOS/library.properties`. Alas, this is not supported

JiriBilek, 2018-02-19:
There is definitely a way how to do it. In boards.txt we could create a new option (similar to the one with optimization (see the key _genericSTM32F103C.menu.opt.osstd_ and following) and use the choice in platform.txt.
Technically, it is possible. I am new to this world, so I would like to leave the decision on more experienced people.

victorpv, 2018-02-20:
It could be added in the boards.txt file as another compile option, similar to the ones to select the optimization level, or use LTO, but that's adding more and more options in boards.txt
If of you make the changes and submit a PR it could be tested.
Between that and FreeRTOS, I get a feeling more people uses FreeRTOS, since there are way more threads about it than about memory usage turning huge, but obviously the best would be to have both benefits.
If sounds like it would be great if compile options could be defined at the library level, but what would happen if you had to include 2 libraries with conflicting options?
At the end of the day, there is no one size fits all.

kolyshkin, 2018-03-06:
btw googling for `freertos -fno-threadsafe-statics` I see many examples where people do use this flag. It would be great to have thread-safe static initialization without the rest of the stuff which is linked in (`__verbose_terminate_handler()` which brings in the whole printf implementation as well as c++ demangling functions).

kolyshkin, 2019-02-20:
@victorpv it would be helpful if someone can find any proof to your statement that adding `-fno-threadsafe-statics` actually breaks RTOS. I tried it, albeit theoretically (see above), but could not find the proof.

Therefore, for now the statement

> The solution could harm if using an RTOS

is just a hypothesis with no substantiation, while https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/444 (which is not merged because of this statement) solves a very practical issue (of not being able to fit a binary into 64K flash).

Can you please take another fresh look at it?

kolyshkin, 2019-02-22:
(from https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/444#issuecomment-466277531)

> hence anyone using RTOS could theoretically have problems next time they compile and run their code.

@rogerclarkmelbourne as I mentioned earlier

> googling for freertos -fno-threadsafe-statics I see many examples where people do use this flag

so this might be "theoretical-only threat" (unless proven otherwise, I mean can anyone show me any existing code that is broken by adding this flag?), while the problem solved by adding this flag is very practical. Just sayin'

AndKe, 2019-02-22:
where do you find the STM32F1/platform.txt ?

andre@loke:~/Arduino$ find . -name platform.txt
./arduino-1.8.8/hardware/arduino/avr/platform.txt
./arduino-1.8.8/hardware/platform.txt

I have also tried to find it in ~/.arduino

victorpv, 2019-02-22:
> @victorpv it would be helpful if someone can find any proof to your statement that adding `-fno-threadsafe-statics` actually breaks RTOS. I tried it, albeit theoretically (see above), but could not find the proof.

I don't remember ever make such statement, can you please let me know in which post I made it?
All I can remember is referencing the warnings from the documentation, and they say using that flag is unsafe on threaded code.
I don't think the flag either always break the code or never breaks the code, form the documentation it seems clear that it can break the code given the right conditions, and not break it if those conditions do not occur.
It's my opinion that's up to the users to decide if they need a safer program when using RTOS, or a smaller program. Including the option for everyone may end up causing crashes that may be hard to track, so I wouldn't do it. I do not own the repo, so I don't make any final decision on what's merged or not, I just state my opinion, so please don't misquote me.

stevstrong, 2019-02-23:
If you installed the files correctly, it should be under
Arduino/hardware/Arduino_STM32/STM32F1

stevstrong, 2019-02-23:
Back to original issue, I think it would make sense to add an "Use RTOS" / "No RTOS" option to IDE menu. That would give place to many other tweaks which are fine for non-OS but are no-go for OS.

kolyshkin, 2019-02-25:
> an "Use RTOS" / "No RTOS" option to IDE menu

This is absolutely not about RTOS, but about code size; the option could be named such as "No threadsafe statics (smaller code)" or "Smaller code (no thread-safe statics").

stevstrong, 2019-02-25:
And i thought you were doubting that this compiler option breaks OS apps. 
I just recommended os menu because then we could have a collection of other os-related compiler options.

victorpv, 2019-02-25:
If added as an option in the menu, we could add a warning in the rtos file, via compiler macro, so the compiler shows a warning if rtos is included while this menu option is set to the opposite. That way at least the user gets a warning.