rogerclarkmelbourne/Arduino_STM32 #299: delay() is missing a yield() call

RickKimball, 2017-06-04:
In newer versions of the Arduino Due core they added a call to yield() to enable the use of the Scheduler library.  The STM32F1 core is missing this.  If you look at the Arduino_Core_STM32F1 code, it has implemented it.   If you aren't using a Scheduler, the yield function becomes a nop and doesn't add anything to the code size.

We need something like this for the Scheduler to work properly
```
extern "C" void yield(void);

void delay(unsigned long ms) {
    uint32 start = millis();
    do {
        yield();
    } while (millis() - start < ms);
}
```
More details in this thread: http://www.stm32duino.com/viewtopic.php?f=3&t=1285#p29123

rogerclarkmelbourne, 2017-06-04:
Thanks Rick

I say the thread, but was not sure what impact adding the call to yield() in delay would have.

Looks like yield is defined as 

static void __empty() {
	// Empty
}
void yield(void) __attribute__ ((weak, alias("__empty")));

in cores/maple/hooks.c

So it would mean additional calls to an empty function, which would have negligible effect on the operation.

So it looks like its safe to implement.

rogerclarkmelbourne, 2017-06-06:
This should be fixed by commit

https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/df78777e98a411036036388cb0ab809f35c6741a

The only additional thing I had to do was include Arduino.h because the the compiler needed the prototype for yield()

closed 2017-06-06 by RickKimball