rogerclarkmelbourne/Arduino_STM32 #300: [STM32F4] [SPI] Compile error in macro SPIx_DMA et al

thetazzbot, 2017-06-06:
I've forked the code and intend to fix this because I need to try to use it.

I am running a Nucleo STM32F411RE board.  I open the Example SPISelfTest sketch and compile it fine under the BluePill board selection.  When attempting to use the F4, it fails with a compile error.  The reason for the error I believe is that the STM32F411RE has 5 SPI ports and the SPI.h file only enumerates three of them when building out the SPIx_ChannelTX and SPIx_ChannelRX defines.

`
C:\Program Files (x86)\Arduino\hardware\STM32GENERIC\STM32\libraries\SPI\src\SPI.cpp:81:17: error: 'SPI4_ChannelTX' was not declared in this scope

    _ChannelTX = SPI4_ChannelTX;

                 ^

C:\Program Files (x86)\Arduino\hardware\STM32GENERIC\STM32\libraries\SPI\src\SPI.cpp:82:17: error: 'SPI4_ChannelRX' was not declared in this scope

    _ChannelRX = SPI4_ChannelRX;

C:\Program Files (x86)\Arduino\hardware\STM32GENERIC\STM32\libraries\SPI\src\SPI.h:114:22: note: in expansion of macro '_SPIx_DMA'

  #define SPIx_DMA(a) _SPIx_DMA(a)

`

I believe the source of the problem is in this list of defines

  #define SPI3_StreamTX 1_Stream5
    #define SPI3_StreamRX 1_Stream0
    #define SPI3_ChannelTX DMA_CHANNEL_0
    #define SPI3_ChannelRX DMA_CHANNEL_0

It stops at 3.  It should probably follow the same paradigm as some other code wherein:
`
#ifdef SPI4
    #define SPI4_StreamTX 1_Stream5
    #define SPI4_StreamRX 1_Stream0
    #define SPI4_ChannelTX DMA_CHANNEL_0
    #define SPI4_ChannelRX DMA_CHANNEL_0
#endif
#ifdef SPI5
    #define SPI5_StreamTX 1_Stream5
    #define SPI5_StreamRX 1_Stream0
    #define SPI5_ChannelTX DMA_CHANNEL_0
    #define SPI5_ChannelRX DMA_CHANNEL_0
#endif
`

Thoughts?

Cheers,
Mark

rogerclarkmelbourne, 2017-06-06:
That bug is not in one of my repo's 

STM32Generic is here

https://github.com/danieleff/STM32GENERIC

closed 2017-06-06 by rogerclarkmelbourne

thetazzbot, 2017-06-06:
Ok thats good to know...