rogerclarkmelbourne/Arduino_STM32 #301: supporting SSD1306 upstream: what are the changes in this repo?

v01d, 2017-06-20:
Hi,
I was using Adafruit's SSD1306 library and then I started using the BluePill board. I realized there is a custom library for STM32 for the SSD1306, but it seems a bit older and has many modifications. I understand the main difference is the use of the hardware I2C. 
I would like to understand what are the minimum set of changes to make the SSD1306 library work, besides the Hardware I2C, in order to have support upstream instead of having this library in this repo.
I tried making a diff but there are many differences, none of which seem important for working.

I'm asking this because using stock Adafruit library I cannot make it work with software nor hardware I2C, so there must be something else going on in this library which makes it work.

rogerclarkmelbourne, 2017-06-20:
That lib was contributed by @madias possibly via the stm32duino.com forum

You would need to contact him about that library, as don't know anything about how it was produced or how it works.

v01d, 2017-06-21:
After lots of trying I managed to get it working, I think it was my mistake. Anyway, after making Adafruit library to use HardWire instead of Wire it works correctly. 

I'm now wondering why it was chosen to use software I2C as default on this framework. Isn't Wire usually hardware implementation? If so, no changes would be needed to any other external library.

rogerclarkmelbourne, 2017-06-21:
When I originally took on maintaining LibMaple, I think hardware I2C was broken.

Subsequently it was fixed by someone (not me) and I accepted a PR for this change

Unfortunately by this time lots of users were using software I2C, so its now a supportabilty problem to swap the default config to hardware.

Please register on the forum as its the best place for these archirectural issues, as I'm just the maintainer, I don't generally make unilateral decisions, I ask the community what they want; or they ask to change things.

I also have to weigh the views of the many vs the views of the few and normally the many will win.

v01d, 2017-06-21:
Got it, thanks.

rogerclarkmelbourne, 2017-07-31:
I've posted this to the forum.

http://stm32duino.com/viewtopic.php?f=53&t=2405

I'll see what the community has to say about it

lacklustrlabs, 2017-12-03:
I just tested the vanilla I2C version of [Adafruit_SSD1306](https://github.com/adafruit/Adafruit_SSD1306) and it worked w/o any problems. 
Does this mean that we can remove the legacy version?
\- If  the upstream SPI version works as well, that is. 

I have a SSD1306 SPI oled somewhere, but I can't find it right now...

(related to #396)

stevstrong, 2019-09-16:
It seems to be solved already, so I close this now.

closed 2019-09-16 by stevstrong