rogerclarkmelbourne/Arduino_STM32 #304: Boards Manager feature request

Testato, 2017-07-04:
One of the biggest improvement on this core will be the possibilities of installing it by standard Arduino Boards Manager.

Last year many people start to ask for a Boards Manager, but the official ST arduino core starting and  it promise to use Boards Manager and we stop to ask it to you.

Now that it is clear that the Official ST core is moving slowly, and that this core will remain for many years active (or i hope forever), I think become important restructure the repo for adding the Boards Manager functionality.

fpistm, 2017-07-05:
Hi @Testato,
FYI, Since last year, all "STM" repos have been merged in one ([Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32) supports all STM32 MCU families). 
It move more quickly, I hope and the number of contributors increase. ;)

Testato, 2017-07-05:
This does not detract from the fact that a board manager for Roger's core is useful and, in my opinion, necessary.
Or has it already been decided that it will be abandoned?

stevstrong, 2017-07-05:
+1
I support the request of Testato.

fpistm, 2017-07-05:
I do not tell that a package is not useful 👍 ;)
I only react on your comment:

> But now that it is clear that the Official ST core is moving slowly,

@rogerclarkmelbourne done/do/will do a great work with several contributors. 🥇

rogerclarkmelbourne, 2017-07-31:
The libmaple still seems to be popular, so I will attempt to maintain it as long as there is enough interest in using it

rogerclarkmelbourne, 2017-08-04:
I believe @ddrown does occasionally update his boards manager package.

I don't personally have time to maintain a boards manager package.

closed 2017-08-04 by rogerclarkmelbourne