rogerclarkmelbourne/Arduino_STM32 #306: F4: Software hangs when USB serial port is not opened and the Tx buffer is full

stevstrong, 2017-07-07:
This happens when the application is sending data over USB serial and the PC has not opened the respective COM port.
This is rather a remainder for me :)

victorpv, 2017-09-06:
Steve, I found that that the usb driver will acknowledge any packet the host sends even if it doesn't fit in the buffer, so it causes the host to send faster than the mcu can read, if the host is fast enough.
I have the code written and tested to correct that. Should I send a PR with that to you or Roger?
And to clarify is only for RX buffer, not related to this TX issue, but since both are about the same driver I thought better comment on this open issue than open a new one.

stevstrong, 2017-09-07:
Yes, please, it would be nice if you could send a PR.

stevstrong, 2017-11-20:
Issue solved by this commit: https://github.com/stevstrong/Arduino_STM32/commit/6d3efff905b1905df69601708c6df6b4f6e48de9

victorpv, 2018-03-06:
https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/452
This PR includes the code so the USB RX code will stop acknowledging the last packet until it can fit in the buffer. No more data loss in that direction. 
Please test and confirms it works for you, I wrote it off an older version, I need to retest it again, but thought on sharing it already in case you get a chance to test before me.
If we can confirm this is all good and merge, we can move on to fix the TX part so it doesn't hang forever.

stevstrong, 2018-04-17:
I close this because the original issue was solved.

closed 2018-04-17 by stevstrong

me21, 2018-06-15:
@stevstrong Please forgive me for hijacking the thread, but there's no private messages on GitHub...
If you'd like to enable issues in your repo, you can do it by clicking "Settings" tab (to the right of "Insights".

stevstrong, 2018-06-16:
Thanks for the hint.