rogerclarkmelbourne/Arduino_STM32 #309: Failed to use ADC

wodiga, 2017-07-09:
When I try to use a STM32ADC example, I got a problem.
Error log like this:
Arduino:1.8.2 (Windows 7), 开发板："Generic STM32F103C series, STM32F103C8 (20k RAM. 64k Flash), Serial, 72Mhz (Normal)"

I:\arduino-1.6.12\arduino-builder -dump-prefs -logger=machine -hardware I:\arduino-1.6.12\hardware -hardware D:\Users\tracis\AppData\Local\Arduino15\packages -tools I:\arduino-1.6.12\tools-builder -tools I:\arduino-1.6.12\hardware\tools\avr -tools D:\Users\tracis\AppData\Local\Arduino15\packages -built-in-libraries I:\arduino-1.6.12\libraries -libraries D:\Users\tracis\Documents\Arduino\libraries -fqbn=Arduino_STM32-master:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=serialMethod,cpu_speed=speed_72mhz -ide-version=10802 -build-path D:\Users\tracis\AppData\Local\Temp\arduino_build_773450 -warnings=default -build-cache D:\Users\tracis\AppData\Local\Temp\arduino_cache_36075 -prefs=build.warn_data_percentage=75 -verbose I:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\libraries\STM32ADC\examples\MultiChannelContinuousConversion\MultiChannelContinuousConversion.ino
I:\arduino-1.6.12\arduino-builder -compile -logger=machine -hardware I:\arduino-1.6.12\hardware -hardware D:\Users\tracis\AppData\Local\Arduino15\packages -tools I:\arduino-1.6.12\tools-builder -tools I:\arduino-1.6.12\hardware\tools\avr -tools D:\Users\tracis\AppData\Local\Arduino15\packages -built-in-libraries I:\arduino-1.6.12\libraries -libraries D:\Users\tracis\Documents\Arduino\libraries -fqbn=Arduino_STM32-master:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=serialMethod,cpu_speed=speed_72mhz -ide-version=10802 -build-path D:\Users\tracis\AppData\Local\Temp\arduino_build_773450 -warnings=default -build-cache D:\Users\tracis\AppData\Local\Temp\arduino_cache_36075 -prefs=build.warn_data_percentage=75 -verbose I:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\libraries\STM32ADC\examples\MultiChannelContinuousConversion\MultiChannelContinuousConversion.ino
Using board 'genericSTM32F103C' from platform in folder: I:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1
Using core 'maple' from platform in folder: I:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1
Detecting libraries used...
"D:\Users\tracis\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10802 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1  -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103C8  -mthumb  -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8  -mthumb  -march=armv7-m -D__STM32F1__ "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple/include" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple/stm32f1/include"                                                           "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/stm32f1" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/usb_lib" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\cores\maple" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\variants\generic_stm32f103c" "D:\Users\tracis\AppData\Local\Temp\arduino_build_773450\sketch\MultiChannelContinuousConversion.ino.cpp" -o "nul"
"D:\Users\tracis\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10802 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1  -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103C8  -mthumb  -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8  -mthumb  -march=armv7-m -D__STM32F1__ "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple/include" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple/stm32f1/include"                                                           "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/stm32f1" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/usb_lib" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\cores\maple" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\variants\generic_stm32f103c" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\libraries\STM32ADC\src" "D:\Users\tracis\AppData\Local\Temp\arduino_build_773450\sketch\MultiChannelContinuousConversion.ino.cpp" -o "nul"
"D:\Users\tracis\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10802 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1  -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103C8  -mthumb  -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8  -mthumb  -march=armv7-m -D__STM32F1__ "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple/include" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple/stm32f1/include"                                                           "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/stm32f1" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/usb_lib" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\cores\maple" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\variants\generic_stm32f103c" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\libraries\STM32ADC\src" "I:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\libraries\STM32ADC\src\STM32ADC.cpp" -o "nul"
"D:\Users\tracis\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10802 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1  -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103C8  -mthumb  -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8  -mthumb  -march=armv7-m -D__STM32F1__ "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple/include" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple/stm32f1/include"                                                           "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/stm32f1" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/usb_lib" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\cores\maple" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\variants\generic_stm32f103c" "-II:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\libraries\STM32ADC\src" "I:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\libraries\STM32ADC\src\utility\util_adc.c" -o "nul"
Generating function prototypes...
"D:\Users\tracis\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -w -x c++ - [... cut ...]
                                        ^

I:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\libraries\STM32ADC\src/STM32ADC.h:156:31: error: 'constexpr' needed for in-class initialization of static data member 'const float STM32ADC::_V25' of non-integral type [-fpermissive]

     static const float _V25 = 1.43; //Volts //1.34 - 1.52

                               ^

使用 1.0  版本的库 STM32ADC 在文件夹： I:\arduino-1.6.12\hardware\Arduino_STM32-master\STM32F1\libraries\STM32ADC 
exit status 1
为开发板 Generic STM32F103C series 编译时出错。

stevstrong, 2017-07-09:
See related forum post: http://www.stm32duino.com/viewtopic.php?f=14&t=2207&p=29875

rogerclarkmelbourne, 2017-07-31:
Close as no reply from the OP within 1 week

closed 2017-07-31 by rogerclarkmelbourne