rogerclarkmelbourne/Arduino_STM32 #310: backslashes in include paths and compile errors, no blink

holla2040, 2017-07-10:
when compiling for STMF407Discovery

The following files have '\' in include paths
STM32F4/variants/discovery_f407/discovery_f4.cpp
STM32F4/cores/maple/libmaple/fsmc.h

/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:48:29: error: 'FSMC_NOE' undeclared here (not in a function)
 const uint8_t fsmc_pins[]= {FSMC_NOE, FSMC_NWE, FSMC_NE1, FSMC_A18,
                             ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:48:39: error: 'FSMC_NWE' undeclared here (not in a function)
 const uint8_t fsmc_pins[]= {FSMC_NOE, FSMC_NWE, FSMC_NE1, FSMC_A18,
                                       ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:48:49: error: 'FSMC_NE1' undeclared here (not in a function)
 const uint8_t fsmc_pins[]= {FSMC_NOE, FSMC_NWE, FSMC_NE1, FSMC_A18,
                                                 ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:48:59: error: 'FSMC_A18' undeclared here (not in a function)
 const uint8_t fsmc_pins[]= {FSMC_NOE, FSMC_NWE, FSMC_NE1, FSMC_A18,
                                                           ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:49:8: error: 'FSMC_D0' undeclared here (not in a function)
        FSMC_D0, FSMC_D1, FSMC_D2, FSMC_D3,
        ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:49:17: error: 'FSMC_D1' undeclared here (not in a function)
        FSMC_D0, FSMC_D1, FSMC_D2, FSMC_D3,
                 ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:49:26: error: 'FSMC_D2' undeclared here (not in a function)
        FSMC_D0, FSMC_D1, FSMC_D2, FSMC_D3,
                          ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:49:35: error: 'FSMC_D3' undeclared here (not in a function)
        FSMC_D0, FSMC_D1, FSMC_D2, FSMC_D3,
                                   ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:50:8: error: 'FSMC_D4' undeclared here (not in a function)
        FSMC_D4, FSMC_D5, FSMC_D6, FSMC_D7,
        ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:50:17: error: 'FSMC_D5' undeclared here (not in a function)
        FSMC_D4, FSMC_D5, FSMC_D6, FSMC_D7,
                 ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:50:26: error: 'FSMC_D6' undeclared here (not in a function)
        FSMC_D4, FSMC_D5, FSMC_D6, FSMC_D7,
                          ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:50:35: error: 'FSMC_D7' undeclared here (not in a function)
        FSMC_D4, FSMC_D5, FSMC_D6, FSMC_D7,
                                   ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:51:8: error: 'FSMC_D8' undeclared here (not in a function)
        FSMC_D8, FSMC_D9, FSMC_D10, FSMC_D11,
        ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:51:17: error: 'FSMC_D9' undeclared here (not in a function)
        FSMC_D8, FSMC_D9, FSMC_D10, FSMC_D11,
                 ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:51:26: error: 'FSMC_D10' undeclared here (not in a function)
        FSMC_D8, FSMC_D9, FSMC_D10, FSMC_D11,
                          ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:51:36: error: 'FSMC_D11' undeclared here (not in a function)
        FSMC_D8, FSMC_D9, FSMC_D10, FSMC_D11,
                                    ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:52:8: error: 'FSMC_D12' undeclared here (not in a function)
        FSMC_D12, FSMC_D13, FSMC_D14, FSMC_D15};
        ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:52:18: error: 'FSMC_D13' undeclared here (not in a function)
        FSMC_D12, FSMC_D13, FSMC_D14, FSMC_D15};
                  ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:52:28: error: 'FSMC_D14' undeclared here (not in a function)
        FSMC_D12, FSMC_D13, FSMC_D14, FSMC_D15};
                            ^
/home/holla/Arduino/hardware/Arduino_STM32/STM32F4/cores/maple/libmaple/fsmc.c:52:38: error: 'FSMC_D15' undeclared here (not in a function)
        FSMC_D12, FSMC_D13, FSMC_D14, FSMC_D15};

stevstrong, 2017-07-11:
The fix is already part of two ongoing PRs, #220 and #305.

holla2040, 2017-07-11:
great, thanks.

closed 2017-07-11 by holla2040