rogerclarkmelbourne/Arduino_STM32 #316: word(...) is missing

VitorBoss, 2017-07-24:
As I told at [http://www.stm32duino.com/viewtopic.php?f=14&t=2142](url) word() functions is missing.

Adding those lines on wirish_math.h
```
#include <stdint.h>

uint16_t makeWord(uint16_t w);
uint16_t makeWord(uint8_t h, uint8_t l);

#define word(...) makeWord(__VA_ARGS__)
```
and those on wirish_math.cpp
```
extern uint16_t makeWord( uint16_t w ) { return w ; }
extern uint16_t makeWord( uint8_t h, uint8_t l ) { return (h << 8) | l ; }
```

rogerclarkmelbourne, 2017-07-31:
Did this work for you?

VitorBoss, 2017-07-31:
Yes, is working fine

rogerclarkmelbourne, 2017-07-31:
Thanks

I'll try adding it to my local copy and will commit as long as it doesnt seem to have any conflicts

rogerclarkmelbourne, 2017-07-31:
Thanks. It should be fixed now.

closed 2017-07-31 by rogerclarkmelbourne

VitorBoss, 2017-08-04:
You forgot STM32F4 folder

rogerclarkmelbourne, 2017-08-04:
The F4 isnt very well supported and is not likely to be,

I will add these functions to the F4, but I'd recommend you use STM's own official core https://github.com/stm32duino for the F4 instead of the libmaple based core.
Because very few people use this core, and I don't actively maintain it now that STM have their official Core.

commit ac27993d (2017-07-31): Added makeWord macro and functions to fix #316