rogerclarkmelbourne/Arduino_STM32 #324: I2c onReceive

scargill, 2017-08-02:
Hi

I'm trying a simple example of using the STM32F103  as an i2cSlave.... using the very latest software as of 2 days ago...

According to the error messages - there are no members called onReceive or onRequest.

What to do next?

I've included WIRE.H and this is where it starts to go wrong

void setup() 
{
    pinMode (LED_PIN, OUTPUT);
  
    // Start the I2C bus as slave
    Wire.begin(OWN_ADDRESS); 
    // Attach a function to trigger when something is received
    Wire.onReceive(receiveEvent);
    // Attach a function to trigger when something is requested
    Wire.onRequest(requestEvent);
}

rogerclarkmelbourne, 2017-08-02:
No one has written I2C Slave for libmaple.

I'd advise you take a look at STM's on official Arduino core to see if this supports it

https://github.com/stm32duino

rogerclarkmelbourne, 2017-08-04:
Closed as there are no plans to support this feature

closed 2017-08-04 by rogerclarkmelbourne

RickKimball, 2017-08-04:
Seems like a reasonable request. It is a supported feature of the Arduino official Wire library.

https://www.arduino.cc/en/Reference/WireOnReceive
https://www.arduino.cc/en/Reference/WireOnRequest

rogerclarkmelbourne, 2017-08-04:
If anyone wants to write this and submit it via a PR, I would be happy to include it.

I don't personally have time to write this, and as STM support this via their official core, there is a way for STM32 users to have this functionality

RickKimball, 2017-08-04:
I'm not suggesting you write it.. I just think it should be on the list for bored contributors : )

RickKimball, 2017-08-04:
Never mind, you already have an open issue for i2c slave

scargill, 2017-08-04:
> <Arduino_STM32@noreply.github.com>
> <author@noreply.github.com>

Yes but right now as far as the cheap board is concerned - at this 
minute those two functions used together do not work - there's a cluge - 
but that's all it is.  I'm now back where I was a year ago - wait for 
further development.

RickKimball, 2017-08-05:
i2c slave isn't implemented, that is the issue not that those 2 functions are implemented.  I had only advocated for this issue because I hadn't seen the issue that describes the real problem.  I was using this as a proxy for that which with that being there, this one doesn't need to be. 

https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/207 .. that placeholder is there when a contributor wants to create a pull request that implements i2c slave.  This is specific to the libmaple version of stm32duino.

@scargill if the new / not yet release version of the ST arduino core were available then you would be set (assuming the i2c slave stuff really works on that core)  It seems like you are happy with your nano stuff i2c slave solution. At some point you can try this stuff later, or roll up your sleeves and make this happen.

scargill, 2017-08-05:
I would like nothing more than to "make it happen" but this particular chipset I know zilch about so the learning curve would likely be too high. My only interest in the chip was that the "Bluepill" board is cheap and hence would make a cheap alternative to the Nano with more pins. In this instance I'll wait for others to finish something they at least are familiar with.