rogerclarkmelbourne/Arduino_STM32 #328: No DFU capable USB device available

drnickyoung, 2017-08-19:
Hi All,

Love the project and the idea, but bit of an issue, might be me missing something. I have got the bootloader on the STM32 and working. Uploaded the blink script and it works fine. Did several times to test out different blink delays, just testing the upload. Had to hit reset a few times, unplug and plug back in a couple but nothing major. Then went to do it again and all I get now is:

> Arduino: 1.8.3 (Mac OS X), Board: "Generic STM32F103C series, STM32F103C8 (20k RAM. 64k Flash), STM32duino bootloader, 72Mhz (Normal), Debug (-g)"
> 
> Sketch uses 17168 bytes (26%) of program storage space. Maximum is 65536 bytes.
> Global variables use 2824 bytes (13%) of dynamic memory, leaving 17656 bytes for local variables. Maximum is 20480 bytes.
> dfu-util 0.9
> 
> dfu-util: Invalid DFU suffix signature
> dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
> Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
> Copyright 2010-2016 Tormod Volden and Stefan Schmidt
> This program is Free Software and has ABSOLUTELY NO WARRANTY
> Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
> 
> dfu-util: No DFU capable USB device available
> An error occurred while uploading the sketch
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.

So the question is what is going on here?

Would appreciate any pointers.
Regards,
Nick

csnol, 2017-08-19:
Yes, I got a same error message.
``` c++
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Deducing device DFU version from functional descriptor length
dfu-util: No DFU capable USB device available
```
BTW,  my dfu-util version is 0.8, How to update to 0.9?

rogerclarkmelbourne, 2017-08-20:
dfu-util is a pain. There are loads of different versions kicking around.

Did you try installing using Homebrew ?
Last time I tried that, it worked for me. But stuff changes all the time, so Homebrew may now install a different version.

csnol, 2017-08-20:
Hi, rogerclarkmelbourne.
My problem's been solved after I updated latest Arduino_STM32 version. Although still got "Invalid DFU suffix signature" . but I can flash in my board use DFU mode (Upload method : STM32duino bootloader.)

Up to yesterday , never got this error use DFU mode on F103 and F107 series board. I don't know what's the reason.
 
4 boards be used to test it.  included: 
>Blue pill   (LED - PC13) ;
>Black pill ( LED - PB12, china version) ;
>Generic STM32F103RCT( LED - PC13) ;
>107VCT(LED - PE0, use STM32F103(PE2) bootloader ) ;

They got a same error message. 
So, I think the reason of the error don't from the hardware.  maybe I updated some other boards or libraies( Arduino_STM32 dependent). and forgot renew this repo.  incompatible case? I can't sure

see below red ****

``` c++
Sketch uses 20864 bytes (7%) of program storage space. Maximum is 262144 bytes.
Global variables use 3704 bytes (7%) of dynamic memory, leaving 45448 bytes for local variables. Maximum is 49152 bytes.
/.../Arduino_STM32-master/tools/macosx/maple_upload cu.usbmodem1411 2 1???:???3
/var/folders/32/***/T/arduino_build*** / RF24L01_RC-STM32F103VCT6-test.ino.bin 
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

*/***dfu-util: Invalid DFU suffix signature****
*/***dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!****
Deducing device DFU version from functional descriptor length
Opening DFU capable USB device...
ID 1???:???3
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 2048
Copying data from PC to DFU device

Download	[                         ]   0%            0 bytes
......
Download	[=========================] 100%        20864 bytes
Download done.
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
Resetting USB to switch back to runtime mode
```

Just suggestion, Could you please issue a json file for this repo?   Everyone can update on time if we add it in preferences.txt.  OK, just suggestion,

Thanks reply, and thanks for your time.

Best regards.

C.S  Nol

drnickyoung, 2017-08-20:
@rogerclarkmelbourne Thank you for the information. Yes is the Blue Pill, sometimes shows on the usb devices sometimes, more often, doesn't show what I'd expect. I did see about the resistor a while ago, but totally slipped my mind to check. Will check later in the week and report back. Thanks for the reminder & help, appreciated.

@csnol I have dfu-utils installed via BREW. Just locate those binaries and copy them over to replace the v0.8 in the STM32Arduino folder.

csnol, 2017-08-20:
Thanks @drnickyoung .
Long time no run BREW,  
```c++
dfu-util 0.9
.....
*/*** dfu-util: Invalid DFU suffix signature ****
*/*** dfu-util: A valid DFU suffix will be required in a future dfu-util release!!! ****
Deducing device DFU version from functional descriptor length
Opening DFU capable USB device...
.....
DFU mode device DFU version 0110
Device returned transfer size 2048
Copying data from PC to DFU device

Download	[                         ]   0%            0 bytes
........
Download	[=========================] 100%        15084 bytes
Download done.
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
Resetting USB to switch back to runtime mode
Waiting for /dev/cu.usbmodem1411 serial...Done

```
Same as 0.8, nothing be changed. 

Best regards.

C.S  Nol

satoshinm, 2017-12-04:
I'm also hitting the "No DFU capable USB device available" error on macOS 10.13 (after having worked around https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/256). Debugging further, Arduino.app is invoking dfu-util like this:

Arduino/hardware/Arduino_STM32/tools/macosx/dfu-util/dfu-util -d 1EAF:0003 -a 2 -D /var/folders/81/3xnrrjrn3hzd1sks0yvxwp1w0000gn/T/arduino_build_216855/Blink.ino.bin -R -R

Device "1EAF:0003" seems incorrect, not sure where it came from. I have Port: "/dev/cu.usbmodem1421 (Maple Mini)" selected in Arduino. Blue pill board, [purchased here](https://www.aliexpress.com/item/Free-Shipping-STM32F103C8T6-ARM-STM32-Minimum-System-Development-Board-Module-Forarduino/32525208361.html). Running `./dfu-util --list`, a different device ID is detected:

```
Found Runtime: [05ac:8296] ver=0061, devnum=4, cfg=1, intf=5, alt=0, name="UNKNOWN", serial="UNKNOWN"
```

If I run dfu-util with the correct device, it gets further. Looks like the IDs are hardcoded? https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/system/Makefile#L108-L111

```make
# Target upload commands
# USB ID for DFU upload -- FIXME: do something smarter with this
BOARD_USB_VENDOR_ID  := 1EAF
BOARD_USB_PRODUCT_ID := 0003
```

and in boards.txt. Am I on the right track by making these changes on my system to find the correct device?:

```diff
--- a/STM32F1/boards.txt
+++ b/STM32F1/boards.txt
@@ -340,8 +340,8 @@ nucleo_f103rb.menu.opt.ogstd.build.flags.ldspecs=
 ###################### Generic STM32F103C ########################################
 
 genericSTM32F103C.name=Generic STM32F103C series
-genericSTM32F103C.vid.0=0x1EAF
-genericSTM32F103C.pid.0=0x0004
+genericSTM32F103C.vid.0=0x05ac
+genericSTM32F103C.pid.0=0x8296
 genericSTM32F103C.build.variant=generic_stm32f103c
 genericSTM32F103C.build.vect=VECT_TAB_ADDR=0x8000000
 genericSTM32F103C.build.core=maple
@@ -374,7 +374,7 @@ genericSTM32F103C.menu.upload_method.DFUUploadMethod.upload.tool=maple_upload
 genericSTM32F103C.menu.upload_method.DFUUploadMethod.build.upload_flags=-DSERIAL_USB -DGENERIC_BOOTLOADER
 genericSTM32F103C.menu.upload_method.DFUUploadMethod.build.vect=VECT_TAB_ADDR=0x8002000
 genericSTM32F103C.menu.upload_method.DFUUploadMethod.build.ldscript=ld/bootloader_20.ld
-genericSTM32F103C.menu.upload_method.DFUUploadMethod.upload.usbID=1EAF:0003
+genericSTM32F103C.menu.upload_method.DFUUploadMethod.upload.usbID=05ac:8296
 genericSTM32F103C.menu.upload_method.DFUUploadMethod.upload.altID=2
 
 genericSTM32F103C.menu.upload_method.serialMethod=Serial
```

rogerclarkmelbourne, 2017-12-04:
I dont think the issue is the USB ID, its just that the bootloader DFU USB device is not being enumerated by the Mac fast enough, as it the bootloader only stays as that device for around 1 second

Several other people have noticed that the newer versions of MacOS are slow to noticed changes to the USB bus, and don't see the DFU device.

I think there are workarounds posted on www.stm32duino.com and probably on one of the issues in github, but possibly the bootloader repo not this repo

satoshinm, 2017-12-04:
I think you're right, if I run [lsusb for Mac](https://github.com/jlhonora/lsusb) in a loop while connecting the board, then it does briefly find `Bus 020 Device 020: ID 1eaf:0003 1eaf Maple 003  Serial: LLM 003` but it quickly disconnects and `Bus 020 Device 021: ID 1eaf:0004 1eaf Maple ` reappears in its place...

Maybe this? https://github.com/rogerclarkmelbourne/STM32duino-bootloader/pull/14 Fixed timing for Mac enumeration and added a variant

rogerclarkmelbourne, 2018-05-23:
Recent updates to the bootloader and core mean that the bootloader should hold in DFU mode if the upload has been triggered by the core being set the Magic number by the IDE

This should resolve the issues on machines which are slow to enumerate the USB

closed 2018-05-23 by rogerclarkmelbourne

TheBeachMaster, 2019-10-02:
> @rogerclarkmelbourne Thank you for the information. Yes is the Blue Pill, sometimes shows on the usb devices sometimes, more often, doesn't show what I'd expect. I did see about the resistor a while [...]

Thanks @drnickyoung  for the suggestion.

Jxid, 2020-03-15:
I am testing an STM32F103 board.
I am using a mac and i had the same problem
"Invalid DFU suffix signature"
when i was trying to upload a sketch.
So the problem solved by pressing the reset button when i press the upload button in Arduino IDE.

watikarosm, 2022-12-14:
I've just join the club of DFU Error.  The solution is "Reset" button.  Press the blue button.

IamDaleon, 2023-07-06:
The rest button doesn't work.... but pulling the cord and jacking back in works, just fine as well!